basler/pypylon #498: Read operation failed, 3791650833, 0xE2000011

smivv, 2022-05-24:
Hello!

Recently I've started to face a problem of camera frame grabbing.
Sometimes grab_result is not succeeded, so I wrapped my code with additional exception which gives me this output:

```
('Grab error: ', 3791650833, 'Read operation failed.')
```

My code looks like this:
```
grab_result = self.camera.RetrieveResult(
    self.grab_timeout, pylon.TimeoutHandling_ThrowException
)

if grab_result.GrabSucceeded():
    frame = grab_result.Array
else:
    raise RuntimeError(
        "Grab error: ", grab_result.ErrorCode, grab_result.ErrorDescription
    )
```

Googling this error code and its hex code `0xE2000011` didn't help me out.
Does anybody know how can I find an actual error corresponding to this issue?

Upd: Found this [thread](https://forum.odroid.com/viewtopic.php?t=29724) which says that there is some problem with USB.
Are there any info about how to fix this?

smivv, 2022-05-25:
Hi @SMA2016a,

OS Ubuntu 18.04

PYLON_VERSION=6.3.0.23157
PYPYLON_VERSION=1.7.4

USB Cable 3m

SMA2016a, 2022-05-26:
try this value

camera.DeviceLinkThroughputLimit.SetValue(300000000)

smivv, 2022-05-26:
Currently it was set to `360000000`, now trying with decreased value `300000000`.

SMA2016a, 2022-05-26:
Using a lower value helps you?

smivv, 2022-05-27:
No, didn't help..

SMA2016a, 2022-05-27:
would you tell me the which camera model are you using and its SN?  or turn to your local distributor to get the firmware update

smivv, 2022-05-27:
Basler acA 1920-155uc, 23155192