basler/pypylon #499: MigrationModeEnable for Emulated cameras

monikascholz, 2022-05-30:
I am trying to get the emulated camera running, which works. However, the emulated camera follows the naming conventions pre SFNC2.0, which means its incompatible with my code which e.g., sets camera.ExposureTime instead of camera.ExposureTimeAbs.

I am trying to enable the migrationMode as briefly mentioned in #10 but I have been unable to do so. Could you help me enable this:
MWE:

```
from pypylon import genicam, pylon
import os

#import cv2
#from libtiff import TIFF

#%% Camera initialization

os.environ["PYLON_CAMEMU"] = "1"
        
  
# Create an instant camera object with the camera device found first.
tl = pylon.TlFactory.GetInstance()
camera = pylon.InstantCamera(tl.CreateFirstDevice())

#camera.TransportLayer.MigrationModeEnable = True
# Register an image event handler that accesses the chunk data.
camera.Open()

#print(dir(camera.TransportLayer.GetNode('MigrationModeEnable')))
print(camera.MigrationModeActive)#.GetVar())
print(dir(camera.GetTLNodeMap().GetNode('MigrationModeEnable')))#.GetNode('MigrationModeEnable')
# Print the model name of the camera.
print("Using device ", camera.GetDeviceInfo().GetModelName()))
```

results in 
```
<Swig Object of type 'Pylon::IBooleanEx *' at 0x7fb8c225f8a0>
Traceback (most recent call last):
  File "/home/scholz_la/Desktop/Code/GUI_Macroscope/TestFunctions/emulation.py", line 24, in <module>
    print(dir(camera.GetTLNodeMap().GetNode('MigrationModeEnable')))#.GetNode('MigrationModeEnable')
  File "/home/scholz_la/.conda/envs/Basler/lib/python3.10/site-packages/pypylon/genicam.py", line 1357, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16139)
```

thiesmoeller, 2022-06-01:
Hi,

MigrationMode is to run new cameras with the old API to support customers with existing code on newer camera models.

We currently don't support your use-case which would be the other way around running a camera with the old feature names with the new API.
See also: https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/what-is-migration-mode-and-how-to-use-it-in-pylon/14592/#

monikascholz, 2022-06-02:
Sure, that works. Do you think we might expect the emulated camera to be updated at some point to follow the standard?

closed 2022-06-02 by monikascholz

thiesmoeller, 2022-06-02:
... it's in the backlog