basler/pypylon #500: Basler Racer SW Line start trigger

MhdKAT, 2022-06-01:
I am trying to implement a SW line start trigger for a basler line scan camera (Basler raL4096-24gm). 
I am trying to follow the user manual  but the functions are not working. Can anyone provide help where to find the corresponding python functions. Thank you!  
[image]

thiesmoeller, 2022-06-01:
Hi @MhdKAT,

pypylon currently has an issue with some of the constants like `TriggerSelector_LineStart`
Workaround is to use the string versions of these values.

So instead of e.g. `camera.Triggerselector.SetValue(TriggerSelector_LineStart)` please write `camera.Triggerselector.SetValue("LineStart")`

Please modify all your other values following the same pattern.

MhdKAT, 2022-06-01:
Thanks for replying @thiesmoeller. I tried what you have suggested but I am still having errors. 
Using device  raL4096-24gm

---------------------------------------------------------------------------
LogicalErrorException                     Traceback (most recent call last)
<ipython-input-74-fea34c4a0451> in <module>()
     10 print("Using device ", cam.GetDeviceInfo().GetModelName())
     11 cam.Open()
---> 12 cam.Triggerselector.SetValue("LineStart")
     13 
     14 

~/.local/lib/python3.6/site-packages/pypylon/pylon.py in __getattr__(self, attribute)
   4301             return object.__getattr__(self, attribute)
   4302         else:
-> 4303             return self.GetNodeMap().GetNode(attribute)
   4304 
   4305     def __setattr__(self, attribute, val):

~/.local/lib/python3.6/site-packages/pypylon/genicam.py in GetNode(self, Name)
   1355 
   1356     def GetNode(self, Name: "GENICAM_NAMESPACE::gcstring const &") -> "GENAPI_NAMESPACE::INode *":
-> 1357         return _genicam.INodeMap_GetNode(self, Name)
   1358 
   1359     def InvalidateNodes(self) -> "void":

LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16139)

closed 2022-06-01 by thiesmoeller