basler/pypylon #501: Frame Rate does not follow the set value

kimdonggyun, 2022-06-02:
I am controlling the camera with Pylon and I wanted to set the Frame rate with the desired value.
I set the Acquisition Frame Rate as 20, but with monitoring of real-time frame rate acquisition, the frame rate was actually 32.
I checked the frame rate set on the camera right before the camera capturing, which always gives the set value (but real capturing is always 32, which might be the maximum FPS of this camera).

Code below:

```
            print("Set value: ", "Height:",cam.Height.GetValue(), "Width:", cam.Width.GetValue(), 
                "Exposuretime:", cam.ExposureTime.GetValue(), "AcquisitionFrameRate:", cam.AcquisitionFrameRate.GetValue())  
            cam.StartGrabbing()
            
            prev_time = 0
            while cam.IsGrabbing():
                try :
                    res = cam.RetrieveResult(1000)
                    new_time_ns = res.TimeStamp # in nano second
                    new_time_s = new_time_ns * 1.e-9 # convert to second
                    livefps = 1/(new_time_s-prev_time)
                    prev_time = new_time_s 

                    print("device name :", cam.DeviceInfo.GetFriendlyName(), "LiveFPS :", livefps)
```

returned:
```
Set value:  Height: 1000 Width: 400 Exposuretime: 8000.0 AcquisitionFrameRate: 20.0
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.46755355036196
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088553394
device name : Basler acA4024-29uc (24258109) LiveFPS : 32.467525088538416
device name : Basler acA4024-29uc (24258110) LiveFPS : 32.467525088553394
```

I do not know which parameter I should change to set the FPS.

*Is there a way to return a timestamp with real date time?
it seems the timestamp embedded on the image always starts 1970-01-01.

I really appreciate any help you can provide.

thiesmoeller, 2022-06-02:
Do you set the enable flag too?

Please see https://docs.baslerweb.com/acquisition-frame-rate.html#setting-the-acquisition-frame-rate
And select your camera model at the top to get exactly the code sample for your device.

You should output the resulting frame rate feature if you want to see the real framerate according to all device settings ( this is referenced from above manual page ) 

As the camera has no realtime clock, it can't output real timestamp. The 1ghz timestamp counter starts at device start time.

You could reference the time from the device with timestamps from your host manually.

kimdonggyun, 2022-06-02:
@thiesmoeller I did not know I have to enable the flag (or set it as Acquasitionframrate to True).
It works now. In case someone wants to get real timestamp, I leave the code below.
```

            prev_time = 0
            timestamp_s_sum = 0
            while cam.IsGrabbing():
                try :
                    res = cam.RetrieveResult(1000)
                    new_time_ns = res.TimeStamp # in nano second
                    new_time_s = new_time_ns * 1.e-9 # convert to second. Timestamp in camera default

                    if timestamp_s_sum == 0:
                        timestamp_s_sum = time.time() - new_time_s # calculate difference between real timestamp and camera timestamp
                    
                    new_time_s = new_time_s + timestamp_s_sum 
                    # sum up difference on camera default timestamp. now new_time_s is real timestamp in seconds
                    livefps = 1/(new_time_s-prev_time)
                    prev_time = new_time_s

```

closed 2022-06-02 by kimdonggyun