basler/pypylon #502: Is there a parameter to see IR light on camera?

kimdonggyun, 2022-06-02:
I wonder if there is a prameter to modify on pylon to make camera see the IR light.
I am using acA4024-29uc.

thiesmoeller, 2022-06-02:
I assume that you mean near IR ( e.g 850nm )

Your camera has an IR cut filter installed.
This is the only way to allow to see proper colours as the color filter array on the pixels is transparent to IR light.

If you want to permanently allow to see NIR light you can remove the filter  according to this app note https://www.baslerweb.com/en/downloads/document-downloads/how-to-remove-the-ir-cut-filter-holder-from-a-basler-ace-camera/#

kimdonggyun, 2022-06-02:
@thiesmoeller thank you so much. I did not know there is a filter installed.

closed 2022-06-02 by kimdonggyun