basler/pypylon #503: Use SW trigger on Basler Racer Line scan camera

MhdKAT, 2022-06-03:
below fixed code and some hints.
* configure the trigger configuration manually. Then you can work along the documentation from docs.baslerweb.com
* you got some frame "by accident", because camera started free-running and in the time between StartGrabbing and TriggerMode=On the pylon grab engine acquired ~10 frames ... so you have a fast camera ;-)
* possibly you found an issue in the SoftwareTriggerConfiguration handler....

```python
import time
from pypylon import pylon
# for recording time limit
import queue
import threading
import numpy as np
import os

converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
img_q = None

#os.environ["PYLON_CAMEMU"] = "1"

class SGNImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, cam, grabResult):
        global img_q, converter
        if grabResult.GrabSucceeded():
            img_q = converter.Convert(grabResult).GetArray()
            # cv2.imshow("Frame",img_q)
            # cv2.waitKey(1)
        # every time you get called, you have a fresh image ...
        # no need to cross check
        print("Image number:", grabResult.ImageNumber)
        return True

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.RegisterImageEventHandler(SGNImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
# configure trigger configuration manually to have full control!
camera.TriggerSelector = "FrameStart"
camera.TriggerSource = "Software"
camera.AcquisitionStatusSelector = "FrameTriggerWait"
# enable trigger mode before StartGrabbing! ... otherwise the camera will start free-running
camera.TriggerMode = "On"
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)

start_rec = int(time.time())
DUR = 60
while int(time.time()) - start_rec <= DUR:
    try:
        if camera.IsGrabbing() and camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
            print("trigger")
            camera.ExecuteSoftwareTrigger()
            print(camera.AcquisitionStatus.Value)
    except Exception as e:
        print(e)

print("End")

camera.Close()

```

_Originally posted by @thiesmoeller in https://github.com/basler/pypylon/issues/287#issuecomment-710942723_
Hi I tried to use this code with a balser racer camera raL4096 but haven't been able to debug it. When I use it as it is I keep on getting "False" from camera.AcquisitionStatus.Value and when I tried to adapt to it to trigger with LineStart instead of FrameStart I kept on having "Waiting for frame trigger ready timed out " . Am I missing any parameter configuration ? and if I want to set multiple SW triggers let's say FrameStart and LineStart how do I Execute those triggers having only one function to do that camera.Execute.SoftwareTrigger(). Thnx!

SMA2016a, 2022-06-04:
<TriggerMode=On the pylon grab engine acquired ~10 frames>  you need to enable triggermode to on before you call Startgrabbing(). So camera will not run in freerun.

thiesmoeller, 2022-06-04:
Hi @MhdKAT,

As a general guidance I would recommend, that You use pylon viewer first to find the working configuration for your use-case ( and experiment quickly with settings) before moving pypylon

MhdKAT, 2022-06-07:
Hi @thiesmoeller. Thnx for your answer so I tried as you said to start experementing with pylonviewer and got the desired functionality. But when I put that into pypylon things don't work as expected : 
-  Desired functionality : I launch the camera once in stream mode and get a full frame after N*LineStart trigger received where N = Height of the image.  
- What I get with pypylon : I get an entire frame each time camera.ExecuteSoftwareTrigger() is executed. 
``` python 
import time
from pypylon import pylon
# for recording time limit
import queue
import threading
import numpy as np
import os

converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
img_q = None

#os.environ["PYLON_CAMEMU"] = "1"

class SGNImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, cam, grabResult):
        global img_q, converter
        if grabResult.GrabSucceeded():
            img_q = converter.Convert(grabResult).GetArray()
            # cv2.imshow("Frame",img_q)
            # cv2.waitKey(1)
        # every time you get called, you have a fresh image ...
        # no need to cross check
        print("Image number:", grabResult.ImageNumber)
        return True

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.RegisterImageEventHandler(SGNImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
# configure trigger configuration manually to have full control!
camera.TriggerSelector = "AcquisitionStart"
camera.TriggerMode = "Off"
camera.TriggerSelector = "FrameStart"
camera.TriggerMode = "Off"
camera.TriggerSelector = "LineStart"
camera.TriggerSource = "Software"
camera.AcquisitionStatusSelector = "FrameTriggerWait"
# enable trigger mode before StartGrabbing! ... otherwise the camera will start free-running
camera.TriggerMode = "On"
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)

start_rec = int(time.time())
DUR = 60
while int(time.time()) - start_rec <= DUR:
    try:
        if camera.IsGrabbing() and camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
            print("trigger")
            camera.ExecuteSoftwareTrigger()
            print(camera.AcquisitionStatus.Value)
    except Exception as e:
        print(e)

print("End")

camera.Close()
```
Am I missing something in the configuration or doing something the wrong way?

MhdKAT, 2022-06-07:
Thnx for replying this is what I get as file : 
'''
# {05D8C294-F295-4dfb-9D01-096BD04049F4}
# GenApi persistence file (version 3.1.0)
# Device = Basler::GigECamera -- Basler generic GigEVision camera interface -- Device version = 3.2.0 -- Product GUID = 1F3C6A72-7842-4edd-9130-E2E90A2058BA -- Product version GUID = F02956CC-AEBB-4EA0-ABF5-003BDC0210C6
GainAuto	Off
GainSelector	All
GainRaw	800
GainSelector	AnalogAll
GainRaw	1
GainSelector	DigitalAll
GainRaw	800
GainSelector	DigitalAll
BlackLevelSelector	All
BlackLevelRaw	50
BlackLevelSelector	All
GammaEnable	0
GammaSelector	User
Gamma	1
PixelFormat	Mono8
ReverseX	0
TestImageSelector	Off
Width	4096
Height	1
OffsetX	0
OffsetY	0
CenterX	0
BinningModeHorizontal	Summing
BinningHorizontal	1
AcquisitionFrameCount	1
TriggerSelector	AcquisitionStart
TriggerMode	Off
TriggerSelector	FrameStart
TriggerMode	Off
TriggerSelector	LineStart
TriggerMode	On
TriggerSelector	LineStart
TriggerSelector	AcquisitionStart
TriggerSource	Software
TriggerSelector	FrameStart
TriggerSource	Line3
TriggerSelector	LineStart
TriggerSource	Software
TriggerSelector	LineStart
TriggerSelector	AcquisitionStart
TriggerActivation	RisingEdge
TriggerSelector	FrameStart
TriggerActivation	RisingEdge
TriggerSelector	LineStart
TriggerActivation	RisingEdge
TriggerSelector	LineStart
TriggerSelector	AcquisitionStart
TriggerDelaySource	Time_us
TriggerSelector	FrameStart
TriggerDelaySource	Time_us
TriggerSelector	LineStart
TriggerSelector	AcquisitionStart
TriggerDelayAbs	0
TriggerSelector	FrameStart
TriggerDelayAbs	0
TriggerSelector	LineStart
ExposureMode	Timed
ExposureAuto	Off
ExposureTimeRaw	320
AcquisitionLineRateAbs	80645.1
FrameTimeoutEnable	0
LineSelector	Line1
LineMode	Input
LineSelector	Line2
LineMode	Input
[... 147 lines cut ...]
ParameterSelector	ExposureOverhead
RemoveLimits	0
ParameterSelector	ExposureOverlapMax
RemoveLimits	0
ParameterSelector	Gain
ExpertFeatureAccessSelector	ExpertFeature1
ExpertFeatureEnable	0
ExpertFeatureAccessSelector	ExpertFeature1
ExpertFeatureEnable	0
ExpertFeatureAccessSelector	ExpertFeature1
ChunkModeActive	0
EventSelector	LineStartOvertrigger
EventNotification	Off
EventSelector	FrameStartOvertrigger
EventNotification	Off
EventSelector	AcquisitionStartOvertrigger
EventNotification	Off
EventSelector	FrameTimeout
EventNotification	Off
EventSelector	ActionLate
EventNotification	Off
EventSelector	EventOverrun
EventNotification	Off
EventSelector	FrameStartOvertrigger
'''

thiesmoeller, 2022-06-07:
okay... the issue with your code is, that you use functionality, which only works for the FrameStartTrigger
```
 if camera.IsGrabbing() and camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
            print("trigger")
            camera.ExecuteSoftwareTrigger()
            print(camera.AcquisitionStatus.Value)
```

the linestart trigger has to be triggered and checked manually

for trigger:
```
camera.TriggerSelector="LineStart"
camera.TriggerSoftware.Execute()
```

To wait for trigger ready you have to poll it manually in a loop
```
camera.AcquisitionStatusSelector = "LineTriggerWait"
camera.AcquisitionStatus.GetValue()
```