basler/pypylon #504: Question about pylon streaming

AndyQiu2234, 2022-06-04:
def take_img(self):
        img = pylon.PylonImage()
        converter = pylon.ImageFormatConverter()

        # converting to opencv bgr format
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        self.camera.StartGrabbing()

        grabResult = self.camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            # Access the image data
            image = converter.Convert(grabResult)
            img = image.GetArray()
        grabResult.Release()

        self.camera.StopGrabbing()

        return img

I wrote a function to export the current pylon camera view as an img and it works for single shot. However, when I tried to set up a timer that repeats this method every second to achieve streaming, the program crushed after running for a few minutes, showing a memory error. I don't actually need to keep all of the images taken during streaming, so I wonder if there is a way to release the memory. Thanks a lot!

SMA2016a, 2022-06-05:
I do not exact root cause of the issue.  But I would approach bit differently.

1) I would use software Trigger. Instead of starting calling camera.StartGrabbing() all the time when an image is needed.
2) image = converter.Convert(grabResult)  does not copy the buffer into your image. 

If you posted a minimum build able sample code that show the issue, I can help you better.

ridvanozdemir, 2022-09-26:
> I do not exact root cause of the issue. But I would approach bit differently.
>
> 1. I would use software Trigger. Instead of starting calling camera.StartGrabbing() all the time when an image is [...]

do you have any sample code for software trigger?

SMA2016a, 2022-09-26:
Please refer the pylon samplecode

https://github.com/basler/pypylon/blob/master/samples/grabstrategies.py

ridvanozdemir, 2022-09-27:
I got this error :

ModuleNotFoundError: No module named 'samples'

SMA2016a, 2022-10-01:
check where you have copied the file imageeventprinter ?  If this is not in subdirectory samples, then you need to adapt 

from samples.imageeventprinter import ImageEventPrinter