basler/pypylon #505: How to set UTC time to the camera

anshkumar, 2022-06-09:
Hi,
I want to set the TimeStamp of the camera to UTC time. My camera is connected to the computer. How to do this ?

thiesmoeller, 2022-06-09:
What is the interface type of your camera ? 

As "setting to UTC" only works natively for GigEVison

anshkumar, 2022-06-09:
I've both usb and Gige.

thiesmoeller, 2022-06-09:
as your "setting to UTC" was not mentioning the clock accuracy that you want to get, there are multiple options:

a) low precision ( usb and gige )
    capture a system timestamp in UTC directly _after_ `cam.RetrieveResult(..)` returns and store it parallel to the image
    This timestamp will be ( t_trigger + (exposure time / readout time / transmission+driver time / userland wakeup ) )

b) very high precision ( gige )
   For the gige cameras you can set up a PTPd on your system and enable IEEE1588/PTP support on your camera ( 
    https://docs.baslerweb.com/precision-time-protocol )
   This timestamp is the trigger timestamp. If you choose to put a GPS synchronized grand master clock  into your network ( e.g. https://www.meinbergglobal.com/english/products/grandmaster-clocks.htm ) you will reach nano second precision on the UTC timestamp.
   Without external clock master: for linux we recommend to follow your distribution documentation to set up the ptpd ( e.g. fedora: 
     https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/servers/Configuring_PTP_Using_ptp4l/ )

anshkumar, 2022-06-09:
For enabling IEEE1588/PTP support on camera, the link gives code in c++. Can you provide it in python please?

thiesmoeller, 2022-06-09:
this is really search'n'replace

// -> # 
;   -> 
true -> True
false -> False
EnumerationName_Value -> "Value"

thiesmoeller, 2022-06-09:
yes. This is what this camera feature realizes.
If your ptpd or ptp4l and your system is configured to sync via ntp to a stratum1 server the camera and your system will run in UTC time  with pretty good precision.

SUNTAO1963, 2022-10-25:
> 

Hi, 
when I try change   c++ 'camera.GevIEEE1588.SetValue(true);' true->Ture in python.
I get error.

```
Using device  acA1300-30gm
An exception occurred.
Traceback (most recent call last):
  File "/home/nano/basler_code/grab.py", line 47, in <module>
    camera.GevIEEE1588.SetValue(True)
  File "/home/nano/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 4545, in __getattr__
    return self.GetNodeMap().GetNode(attribute)
  File "/home/nano/.local/lib/python3.6/site-packages/pypylon/genicam.py", line 1473, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16615)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nano/basler_code/grab.py", line 69, in <module>
    print(e.GetDescription())
AttributeError: 'LogicalErrorException' object has no attribute 'GetDescription'
```

How can I use python to set the TimeStamp of the camera to UTC time?

thiesmoeller, 2022-10-25:
If you can't access this feature on pylon viewer neither, it is missing from your camera model.

As your camera model is an acA1300-30gm ace classic it might be, that there is no PTP support available.
Check with your support contact, if a firmware update exists.