basler/pypylon #507: acA3800-14uc (USB) only capture grey pictures

devimauz, 2022-06-13:
Hi,
my problem is with the camera it only takes gray pictures. With other camera i can grab color pictures. With PylonViewer it works.
My code is:```
camera = 
pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.PixelFormat.SetValue("BayerBG8")
camera.PixelColorFilter.Setvalue = "BayerBG"
camera.Width = 3840
camera.Height = 2748
camera.OffsetX = 0
camera.OffsetY = 0
camera.ExposureTime = 29435
```

Thanks for the help!

thiesmoeller, 2022-06-13:
You configure it to capture Bayer RAW data.
So this is as expected. As no color interpolation takes place in the camera.

As pylon viewer is applying  color interpolation automagically you see a color image.
It uses the ImageFormatConverter class.
Sample code in: https://github.com/basler/pypylon/blob/master/samples/utilityimageformatconverter.py

devimauz, 2022-06-13:
Thank You!

Thies Möller ***@***.***> ezt írta (időpont: 2022. jún. 13.,
H 17:30):

closed 2022-06-16 by thiesmoeller