basler/pypylon #508: Creating a .exe file with Pypylon module

virginiaroasters, 2022-06-19:
[image]

Hello, I'm trying to convert a python script to a .exe file using Auto Py to Exe and I'm getting the following error regarding the Pylon module. I installed pypylon from Anaconda using Venv.

thiesmoeller, 2022-06-19:
We recommend to use pyinstaller.

EtalumaSupport, 2022-06-21:
I am having the same issue with code compiled using pyinstaller. What's frustrating is that the same .spec file previously worked.
[image]

thiesmoeller, 2022-06-21:
the current pyinstaller distribution contains a pypylon hook file, but this hook file on linux has an issue with packaging part of the USB driver libs.

you have to reference them by 

put it into a file `hook-pypylon.py` and reference the folder of this file with  `--additional-hooks-dir <path_to_folder_of_fixed_hook_file>` on pyinstaller call

```python
# ------------------------------------------------------------------
# Copyright (c) 2020 PyInstaller Development Team.
#
# This file is distributed under the terms of the GNU General Public
# License (version 2.0 or later).
#
# The full license is available in LICENSE.GPL.txt, distributed with
# this software.
#
# SPDX-License-Identifier: GPL-2.0-or-later
# ------------------------------------------------------------------

# PyPylon is a tricky library to bundle. It encapsulates the pylon C++ SDK inside
# it with modified library references to make the module relocatable.
# PyInstaller is able to find those libraries and preserve the linkage for almost
# all of them. However - there is an additional linking step happening at runtime,
# when the library is creating the transport layer for the camera. This linking
# will fail with the library files modified by pyinstaller.
# As the module is already relocatable, we circumvent this issue by bundling
# pypylon as-is - for pyinstaller we treat the shared library files as just data.

import os
import PyInstaller.utils.hooks as hooks

# Patterns of dynamic library filenames that might be bundled with some installed Python packages.
# extended to cover the unsually named pylon-libusb
hooks.PY_DYLIB_PATTERNS = [
    '*.dll',
    '*.dylib',
    'lib*.so',
    'pylon*.so',
]

# Collect dynamic libs as data (to prevent pyinstaller from modifying them)
datas = hooks.collect_dynamic_libs('pypylon')

# Collect data files, looking for pypylon/pylonCXP/bin/ProducerCXP.cti, but other files may also be needed
datas += hooks.collect_data_files('pypylon')

# Exclude the C++-extensions from automatic search, add them manually as data files
# their dependencies were already handled with collect_dynamic_libs
excludedimports = ['pypylon._pylon', 'pypylon._genicam']
for filename, module in hooks.collect_all('pypylon._pylon')[0]:
    print(filename, module)
    if (os.path.basename(filename).startswith('_pylon.')
            or os.path.basename(filename).startswith('_genicam.')):
        datas += [(filename, module)]
        
```

kotai2003, 2022-07-21:
Hi, Here is my spec. file. It seems to be excessive. but it works clearly.

`import pypylon
import pathlib
from pypylon import pylon
from pypylon import genicam
from pypylon import _genicam
from pypylon import _pylon

pypylon_dir = pathlib.Path(pypylon.__file__).parent
pypylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('*.dll')]
pypylon_pyds = [(str(dll), '.') for dll in pypylon_dir.glob('*.pyd')]

_binaries = list()
_binaries.extend(pypylon_dlls)
_binaries.extend(pypylon_pyds)

_pathex = list()
_pathex.append(str(pypylon_dir))

_hiddenimports = list()
_hiddenimports.extend(['pypylon', 'pypylon.pylon', 'pypylon.genicam', 'pypylon._pylon', 'pypylon._genicam'])

a = Analysis(
    ['your_apps.py'],
    pathex=_pathex,
    binaries=_binaries,
    datas=[],
    hiddenimports=_hiddenimports,
   ...
)
`

iwaniec, 2022-07-25:
[kotai2003] It worked! Thank you! At least for my test code that only includes pylon and not kivy. For reference, I am working on the project listed by EtalumaSupport (https://github.com/EtalumaSupport/LumaViewPro). 

`# -*- mode: python ; coding: utf-8 -*-

block_cipher = None

import pypylon
import pathlib
from pypylon import pylon
from pypylon import genicam
from pypylon import _genicam
from pypylon import _pylon

pypylon_dir = pathlib.Path(pypylon.__file__).parent
pypylon_dlls = [(str(dll), '.') for dll in pypylon_dir.glob('.dll')]
pypylon_pyds = [(str(dll), '.') for dll in pypylon_dir.glob('.pyd')]

_binaries = list()
_binaries.extend(pypylon_dlls)
_binaries.extend(pypylon_pyds)

_pathex = list()
_pathex.append(str(pypylon_dir))

_hiddenimports = list()
_hiddenimports.extend(['pypylon', 'pypylon.pylon', 'pypylon.genicam', 'pypylon._pylon', 'pypylon._genicam'])

a = Analysis(
    ['script_example.py'],
    pathex=_pathex,
    binaries=_binaries,
    datas=[],
    hiddenimports=_hiddenimports,
    hookspath=[],
    hooksconfig={},
    runtime_hooks=[],
    excludes=[],
    win_no_prefer_redirects=False,
    win_private_assemblies=False,
    cipher=block_cipher,
    noarchive=False,
)
pyz = PYZ(a.pure, a.zipped_data, cipher=block_cipher)

exe = EXE(
    pyz,
    a.scripts,
    [],
    exclude_binaries=True,
    name='script_example',
    debug=False,
    bootloader_ignore_signals=False,
    strip=False,
    upx=True,
    console=True,
    disable_windowed_traceback=False,
    argv_emulation=False,
    target_arch=None,
    codesign_identity=None,
    entitlements_file=None,
)
coll = COLLECT(
    exe,
    a.binaries,
    a.zipfiles,
    a.datas,
    strip=False,
    upx=True,
    upx_exclude=[],
    name='script_example',
)
`