basler/pypylon #509: Pylon Error: cameras have been physically removed

MehdiiHs, 2022-06-22:
I'm working with a jetson nano board and a Basler Ace camera, after my program worked for a while it shows an error and says "camera have been physically removed" and I can't get frames from the camera anymore. this error happens randomly not always. what caused this error and how can I fix this error?

SMA2016a, 2022-06-25:
what interface? USB or GigE?

USB : cable quality , autosuspend mode etc..
GigE : Netqwwork overloaded?

hutomosaleh, 2022-07-15:
> what interface? USB or GigE?
> 
> USB : cable quality , autosuspend mode etc..
> GigE : Netqwwork overloaded?

Hi, i'm experiencing this error too, I'm using GigE. Can you elaborate on the network overload problem?

thiesmoeller, 2022-07-18:
Depending in your camera model, Basler GigE cameras can fully saturate the network connection.
This could result in issues if your host is not fast enough to handle the amount of data received.

If you start the pylon viewer on your system you will find a menu item under Tools with a bandwidth manager which allows to find optimal settings for your system

niilona, 2022-08-30:
It's a hard to get Basler camera in use after it's disconnected/connected again.

bnascimento, 2023-02-28:
We face this issue as well: just open pylonviewer on a jetson orin I randomly get the following
`[2023-02-28 16:08:13.856][Info] Basler acA1920-40gc (24553637): "Basler acA1920-40gc (24553637)" has been removed.
[2023-02-28 16:08:23.857][Info] Basler acA1920-40gc (24553637): "Basler acA1920-40gc (24553637)" has been detected.
`
It cant be network load, as sometimes the camera itself is not being used at all.

alorcamo, 2023-03-29:
Hi, I am getting this error too. 

Setup: 2x Basler a2A2590-22gcBAS connected to an Intel I350-T4 ports B and C. 

I get in one of them also: "Error:  3774873620 The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.". 

I run the Bandwidth manager to optimize it. Jumbo packets are set to max of 9014 Bytes. 

I interestingly enough, sometimes it works fine, sometimes not....

PaVaNTrIpAtHi, 2023-05-22:
Same error here. Was anyone able to resolve this?

EvanBrownVTM, 2023-09-18:
Same issue. Any updates?

maschbauerkl, 2023-11-28:
I'm getting the same error. Sometimes, it's just not grabbing and after a while it's throwing this error. The strange thing is: The same setting (grabbing images as a background task inside a docker container) is working fine on my 12 year old laptop, but is 100% of the time crashing on a state of the art desktop pc with way more resources.

Rama-Uskamolla, 2026-09-16:
Anyone resolved this issue?