basler/pypylon #510: Using USB3 cameras with a fiber-optic extender on Jetson Nano+Ubuntu 18.04

milancurcic, 2022-06-22:
Hi,

We're hitting an interesting issue when using a USB3 camera with a fiber-optic extender.

## Setup

* Board: NVIDIA Jetson Nano
* OS: Ubuntu 18.04 64-bit
* Camera: [Basler aca2440-35um](https://docs.baslerweb.com/aca2440-35um.html)
* Pylon SDK: 6.2.0
* Fiber-optic extender: [Transwan Mini USB 3.0 fiber optic extender](https://www.amazon.com/Transwan-Extender-Single-Mode-Multi-Mode-Supports/dp/B08JGFSJBT)

## Issue

* We are able to open the camera, query its settings, and update its settings, both via pypylon and Pylon Viewer.
* When we try to grab images, either via pypylon or Pylon Viewer, the camera hangs and returns with the following error:
```
 File "/path/to/venv/lib/python3.8/site-packages/pypylon/pylon.py", line 3405, in StartGrabbing
    return _pylon.InstantCamera_StartGrabbing(self, *args)
_genicam.RuntimeException: PrepareGrab failed for device '2676:ba05:2:1:3'. Error: 'The requested operation timed out.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 262)
```
When in Pylon Viewer, the error message is the same, but it appears in a popup instead.

Through some experimentation, we have isolated the fiber-optic extender + Linux (or Jetson) as the culprit:

1. If we remove the fiber-optic extender from the setup, i.e. if we connect with a short USB3 cable, the Jetson Nano system works without issues and streams at 35 fps as expected.
2. If we use the fiber-optic extender with the same camera but with a x86-64 Windows 10 computer, there are no issues and we get a 35 fps stream.

So:

| | USB3 cable | USB3 cable + fiber-optic extender | 
|---|---|---|
| x86-64 Windows 10 | OK | OK
| Jetson Nano + Ubuntu 18.04 | OK | **Error** (as described above)

We contacted the vendor and they confirmed that no additional drivers are needed for Ubuntu 18.04 and fiber-optic extender.

We set all relevant USB parameters on Linux, as recommended by the Pylon SDK guide (`usbfs_memory_mb` is 1000, and `cam.StreamGrabber.MaxTransferSize` is 4194304).

Any help with how we can further diagnose this, or if anybody has tried using a fiber-optic extender in a similar setup and found a working solution, I'd greatly appreciate it.

thiesmoeller, 2022-06-22:
@milancurcic please ask your vendor explicitly for Jetson nano.

If you just state Ubuntu ... I assume they answer for a PC setup..

It is typically not an issue with the OS or driver but with the real silicon implementation of the host controller.

milancurcic, 2022-06-22:
Thanks, we will. We already asked about Ubuntu on ARM, and the answer was of the generic kind, "all that the system needs to support is USB3.0 and the fiber-optic extender should work".