basler/pypylon #514: Pypylon connot import name _pylon

sulaimansiddiqi233, 2022-07-09:
Hello to every one .
I am using Basler Camera in my project is is working in pycharm and I create the bat file it is working also there. but the problem is that I want to convert my code from .py to .exe then I get the error. as you can see in attachment file. could help me out. thank you.
I am using window system. 64 bit . python 3.8 65 bit and pypylon 1.8.0 version
[image]

thiesmoeller, 2022-07-09:
What version of pyinstaller do you use ?

sulaimansiddiqi233, 2022-07-09:
pyinstaller 5.2

thiesmoeller, 2022-07-09:
Can reproduce. 
It is longer time ago, that I tested pyinstaller on windows.
The pypylon hook ( part of pyinstaller ) should normally take care to collect all required libraries.
Why this is broken now with the very current pyinstaller 5.2 version has to be checked....
As we haven't changed the library layout for windows

sulaimansiddiqi233, 2022-07-09:
in which version of pyinstaller it works correctly . I will try that one

thiesmoeller, 2022-07-09:
As one check:
Are you calling pyinstaller from a virtualenv?

sulaimansiddiqi233, 2022-07-10:
no my code is in pycharm and it is connected with python I am call the pyinstaller directly from cmd with all libraries .

thiesmoeller, 2022-07-11:
With "ran your code" you mean that you used above code as the patched pyinstaller hook file?

thiesmoeller, 2022-07-11:
But you show the output of the python console....

Not the terminal ...

How do you call pyinstaller?

sulaimansiddiqi233, 2022-07-11:
İn cmd : pyinstaller code.py

thiesmoeller, 2022-07-11:
Save my above code into a file hook-pypylon.py in a directory .\hooks

`pyinstaller --additional-hook-dir .\hooks  <your_main_script.py>`

This way you override the buggy hook file of pyinstaller

sulaimansiddiqi233, 2022-07-11:
I added the code but it given the below error/ Sent from Mail for Windows From: Thies MöllerSent: Monday, July 11, 2022 7:13 PMTo: basler/pypylonCc: sulaimansiddiqi233; AuthorSubject: Re: [basler/pypylon] Pypylon connot import name _pylon (Issue #514) Save my above code into a file hook-pypylon.py in a directory .\hookspyinstaller --additional-hook-dir .\hooks <your_main_script.py>This way you override the buggy hook file of pyinstaller

thiesmoeller, 2022-07-11:
Could you please comment from the webinterface. Mail replies are unreadable

sulaimansiddiqi233, 2022-07-11:
i added the code after i run the . i got the error
[image]

thiesmoeller, 2022-07-11:
Sorry typo...

--additional-hooks-dir ....

sulaimansiddiqi233, 2022-07-12:
It works Sir . thanks a lot.

closed 2022-07-12 by thiesmoeller