basler/pypylon #515: How to add pylon viewer vtool recipe file in to python.

sulaimansiddiqi233, 2022-07-12:
hello sir. we just testing the pylon viewer vtools for barcode reading from basler camera . we just add the the vtools objects and configurate them. we can get the result in pylon viewer application. so how we can export the result or how can we integrate the vtools in python code. 
[image]
thank you.

sulaimansiddiqi233, 2022-07-13:
thank you sir . so could me send a simple code of C++ that I can load recipe file of vtools and save the result in a file.

m-binev, 2022-07-13:
@sulaimansiddiqi233 Hi, well, you have to either order a free evaluation license to be able to run vTools in API (with limitted execution time and for limitted period), or buy a runtime license (no limitations) for the vTool(s) you want to use. Then I recommend creating your own recipe for your camera and your use case and refer to e.g. the barcode SDK sample part of the pylon C++ Data Processing SDK - it should be a piece of cake to get things running then.  For ordering an evaluation license, check out that page: https://www.baslerweb.com/en/products/basler-pylon-vtools/pylon-vtools-inquiry/

m-binev, 2022-07-13:
@sulaimansiddiqi233 Hi, you cannot run the vTools/recipes in the API with a demo license. The demo license is only for testing the pylon Viewer/Workbench. Please refer to my previous posting. Cheers

sulaimansiddiqi233, 2022-07-13:
ok thank. we can not afford the license for now. we will wait for the python connection API then we can try it. could you send us updates if the python recipe API activated. thank you.

SMA2016a, 2022-08-22:
@sulaimansiddiqi233 : 
you dodnt need to wait until pypylon version that supports natively vTools. 
if you wanted use pylon vTools in pypylon right now, you can write a small wrapper to convert  np::ndarray to pyloyn image and pass the pylon image to  vTools/recipes. 
After purchasing a valid license, turn to Basler support team, to get sample code for the wrapper.

sulaimansiddiqi233, 2022-11-03:
hello sir. 

we just buy the pylon barcode reader basic with the USB dongle. so now how I can run it with c++ code and make wrapper to send the data from c++ to python.

SMA2016a, 2022-11-04:
OK, fine. Send me your email ID via private message, I will write the wrapper

sulaimansiddiqi233, 2022-11-04:
> OK, fine. Send me your email ID via private message, I will write the wrapper
@SMA2016a  thank you for your replay. 
Email Id : sulaimansiddiqi8@gmail.com
best regards

sulaimansiddiqi233, 2022-11-07:
OK, fine. Send me your email ID via private message, I will write the wrapper
@SMA2016a thank you for your replay.
Email Id : sulaimansiddiqi8@gmail.com
best regards

sulaimansiddiqi233, 2022-11-07:
@SMA2016a thank you for your replay.
Email Id : sulaimansiddiqi8@gmail.com
best regards

SMA2016a, 2022-11-07:
Thanks for sending your email. I am working on writing the wrapper. Get back to you soon.

sulaimansiddiqi233, 2022-11-14:
@SMA2016a  hello Sir/Madam . Is there any updates about wrapper. 

best regards

Mechamon, 2022-11-15:
@SMA2016a I have a similar project that needs a sample wrapper to be written from C++ to show on python, and it uses the measurement vtool. I have an evaluation license that allows the vtool to be used for the application code. can I get a sample code sent to me as well. Thanks.

SMA2016a, 2022-11-15:
sure, it is near to finish. Will you also send me an PN, please.

sulaimansiddiqi233, 2022-11-15:
@SMA2016a  Hello , 
PN? do you mean phone number ?

Mechamon, 2022-11-16:
@SMA2016a my email is nizwiz25@gmail.com thanks. look forward to your response.

sulaimansiddiqi233, 2022-11-16:
@SMA2016a  hello . I could not find your email. so can you send an email to me. then we can continue from there. 
email id .: sulaimansiddiqi8@gmail.com
best regards

sulaimansiddiqi233, 2022-11-20:
@SMA2016a  thank you for your support we just got the code but we could not run it with MinWG or visual . so could we arrange an online meeting to check the problem and solve it . thank you.

sulaimansiddiqi, 2023-07-28:
Hello dear @SMA2016a @m-binev . is there any update about pylon vtools that we connected to python directly . we using the vtools for barcode reading and connected program from c++ to python . is there any direct library so please update us. 
thank you.

agvdndor, 2023-11-03:
Thanks for the work! I'm also interested in the python wrapper for vtools. Any chance we can help contribute to pypylon so it can be a part of the next release?

m-binev, 2023-11-03:
@sulaimansiddiqi @agvdndor @SMA2016a Hi all, I'm glad to share the info, that there are very good chances we are launching the pylon vTools in pypylon next week! This will be the reason to increase the major version to 3.0. Stay tuned))

m-binev, 2023-11-12:
@sulaimansiddiqi @agvdndor @SMA2016a Hi all, it's done! Please use pypylon 3.0.0 for using all latest pylon vTools as available in the last pylon 7.4.0 release. Please check the ReadMe and for using licensable vTools, you'll need either an Evaluation or a Runtime license, of course. Cheers