basler/pypylon #519: Increase Frame Rate/Transfer Speed for saving .png-images (at best without compression)

visionbyalex, 2022-07-26:
Hello everybody!

I have following problem: I want to save images as png but the frame rate/transfer speed and the saving process are really slow compared to .tiff and .jpeg. I use a acA5472-17uc (USB3) triggered by hardware every 1000 ms. The saved frames are correct but the saving process' duration is enormously long. The next step would be to increase the acquiring speed to 100 ms and I am afraid to lose some frames using png format. Normally I would just change the format form from png to jpeg or tiff but after capturing the images an processing algorithm needs png format. I figured out that the png saving routine compresses the captured frames and that's why the saving process is so slow. Is there any solution to save png images without compression or to increase the saving? 

This is the used code:

```
from pypylon import pylon
from PIL import Image
import time

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()
cam.ExposureTime.SetValue(5000)
cam.PixelFormat.SetValue("RGB8")
cam.TriggerSelector.SetValue('FrameStart')
cam.AcquisitionMode.SetValue('Continuous')
cam.TriggerMode.SetValue('On')
cam.TriggerSource.SetValue('Line4')
cam.TriggerActivation.SetValue('RisingEdge')
cam.MaxNumBuffer = 5
cam.StreamGrabber.MaxTransferSize = 4 * 1024 * 1024

tlf = pylon.TlFactory.GetInstance()
devices = tlf.EnumerateDevices()
for device in devices:
    print(device.GetFriendlyName() + " online")
# print(cam.PixelFormat.Symbolics)
print("--------------------------------------")

while True:
    inp = input("Press 1 to zero system, 2 for Calibration, 3 for stop a go, 4 for continuous Mode: ")

    if inp == '1':
        ser.write(str.encode(inp))
        print("Zero system")
        readfromard()

    elif inp == '2':
        ser.write(str.encode(inp))
        print("System Calibration")

        cam.StartGrabbing()

        for m in range(1, 5):
            with cam.RetrieveResult(2000) as res:
                t1 = time.time()
                img.AttachGrabResultBuffer(res)
                filename = 'cal1_%d.tif' % m
                img.Save(pylon.ImageFileFormat_Tiff, output_path + filename)
                t2 = time.time()
                img.Release()
                print('fps: %.2f' % (1 / (t2 - t1)), 'bandwidth: %.2f MB/s' % (image_size / 1024 ** 2 / (t2 - t1)))

        readfromard()
        cam.StopGrabbing()
        cam.Close()
        break
```

And the following images show the different saving speeds and frame rates:

- Tiff:

[image]

- Jpeg:

[image]

- Png:

[image]

As you can see the saving routine using tiff is about 35 times faster than the png saving routine

thiesmoeller, 2022-07-26:
PIL's PNG encoding has a really low performance.

Either save in tiff (which is really just saving the raw buffer + some header ) and then use some tool after grabbing for conversion. E.g. imagemagick convert.

Or try using e.g. the python wrapper around high performance PNG encode fpng

https://github.com/qrmt/fpng-python

visionbyalex, 2022-08-04:
Alright, thank you. I will try it with fpng.

closed 2022-08-04 by visionbyalex