basler/pypylon #522: Cannot import pypylon modules on jetson xavier

Zillibub, 2022-08-02:
Hello, 

Pylon version: 7.1.0

I build module with the following command: `pip install git+https://github.com/basler/pypylon.git` in venv

But when I try to import module, it returns the following error: 
```
from pypylon import pylon
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/inference-env/lib/python3.8/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: libpylonbase.so.7.1: cannot open shared object file: No such file or directory
```

thiesmoeller, 2022-08-02:
As a quick workaround install latest  pylon 6.x version.
7.1 support hasn't been integrated yet

Zillibub, 2022-08-02:
Worked like a charm, thanks!

tonynajjar, 2023-09-28:
Any update if pypylon supports Pylon versions >7.*?