basler/pypylon #523: Continuous Acquisition mode

varagantis, 2022-08-03:
Hello,

I need to run pylon Continuous acquisition mode (Video Mode) in the pylon GUI via pypylon. Is there any function or code sample I can reference.

Thank you,
Srikar. V

thiesmoeller, 2022-08-03:
What do you mean by 'pylon GUI' ?

varagantis, 2022-08-03:
I mean pylon viewer.

thiesmoeller, 2022-08-04:
If you only require the display portion of pylon viewer you can replace this with a Qt ( PyQt or PySide ) version.

Or which functionality of Pylon Viewer, do you want to keep, while running image processing in pypylon?

varagantis, 2022-08-04:
So, I'm kind of confused with which function within the pypylon would help me start the continuous acquisition mode. I found one from registration mode, but wasn't quite sure how would I then use this mode to open pylon viewer and show the video.

thiesmoeller, 2022-08-10:
The connection of an application to the camera is point-to-point.
So e.g. if you open your camera in Basler pylon viewer, you can't open it a second time in a pypylon python script.

> kind of confused with which function within the pypylon would help me start the continuous acquisition mode

So you only want to automatically start acquisition in the pylon viewer?

Or do you want to do some real image processing in pypylon and have only an additional component to display the video?

varagantis, 2022-08-10:
Oh yeah, Thank you for reminding me of that fact. 

In this case I think I would like an additional component to display the video.