basler/pypylon #524: Image width limitation

nathanaelvw, 2022-08-04:
Hi, I was wondering if it is possible to create an image frame longer (in terms of height) than the camera's limit.

PolymaTh-EE98, 2022-08-04:
The title says width but question seems to be about height...
https://docs.baslerweb.com/image-roi.html#changing-position-and-size-of-an-image-roi

I assume this is regards to linescan cameras?
For me there is an issue trying to set the height to something longer than what is in the camera. So I go into the camera (or pfs file) and make it longer than I will need (or the cameras maximum).  Then when my application launches I set it to the correct value. 

You will also notice in pylon viewer the ability to change the height while it is acquiring is disabled.  So try something like this:
                       #  CameraHeight is your calculated value for the image height
                        camera.AcquisitionStop.Execute()
                        camera.TLParamsLocked = False
                        camera.Height = int(CameraHeight)
                        camera.TLParamsLocked = True
                        camera.AcquisitionStart.Execute()

nathanaelvw, 2022-08-05:
Correct, sorry for the inconvenience. In this case, what is the reason for the height limit? If I am not mistaken, it is not a hardware limitation(?) Thus, unlocking the limit is not harmful in any way(?)

PolymaTh-EE98, 2022-08-05:
To me it seems there are 2 max heights and one in my opinion is a bug.  If you go in the pylon viewer you can look at the device properties to see the Max Height.  You cannot exceed this no matter what. 
camera.HeightMax.GetValue()
I currently have a raL2048-48gm plugged in and my device max height is 12288.

I tried to say above if you go in the AOI controls and set the height to something less than device max height it seems to set a different limit to the camera height less than the max of the device. 
camera.Height.GetValue()    --  this is parameter you changing with the code above but like I said it seems you cannot change it to something more than what is in the pfs file (or if you are not using that what was last saved to the camera)

nathanaelvw, 2022-08-08:
My maximum height value based on camera.HeightMax.GetValue() is 12288, but again, using:

camera.AcquisitionStop.Execute()
camera.TLParamsLocked = False
camera.Height = int(CameraHeight)
camera.TLParamsLocked = True
camera.AcquisitionStart.Execute()

only gave me the maximum of 7743. At the same time, when I tested the code out of my main code (just to test the camera instead of running my complete application), the maximum height is only 3750.

nathanaelvw, 2022-08-08:
from pypylon import pylon

tl_factory = pylon.TlFactory.GetInstance()
camera = pylon.InstantCamera()
camera.Attach(tl_factory.CreateFirstDevice())

camera.Open()

camera.MaxNumBuffer = 20

camera.BlackLevelRaw.SetValue(50)
camera.ExposureTimeAbs.SetValue(98)
camera.GainRaw.SetValue(2047)
camera.CenterX.SetValue(True)
camera.Width.SetValue(7743)

camera.AcquisitionStop.Execute()
camera.TLParamsLocked = False
camera.Height = int(7743)
camera.TLParamsLocked = True
# print(camera.HeightMax.GetValue())
print(camera.Height.GetValue())

camera.Close()

camera.DetachDevice()

thiesmoeller, 2022-08-08:
Maximum height of a linescan camera is given by amount of internal memory to guarantee transfer to the host.

Just stitch multiple shorter frames together to get larger height.

nathanaelvw, 2022-08-09:
I was also thinking of doing the same, but isn't there going to be a delay before the camera can capture the next frame?

nathanaelvw, 2022-08-09:
I just tried the idea, and it works. There is no delay whatsoever. Thank you guys for the help!

closed 2022-08-09 by nathanaelvw