basler/pypylon #525: How to configure camera bandwidth with python ?

MehmetOKUYAR, 2022-08-11:
ı want to configure each camera bandwidth on the python script. how to ı configure, do you know ? ı can configure it on pylon viewer but ı want to set it with python

thiesmoeller, 2022-08-12:
You have to set the exact same features as in the viewer.

Please check our partner repository https://github.com/basler/pypylon-samples for background information.

MehmetOKUYAR, 2022-08-13:
@thiesmoeller  ı didn't find what ı want in these samples

thiesmoeller, 2022-08-14:
So best describe what you want.
Which features of your camera (model?)  do you want to configure

The examples in https://github.com/basler/pypylon-samples demonstrate how to access any Feature exactly like in the c++ sample code snippets in the Basler Produkt documentation

MehmetOKUYAR, 2022-08-14:
[image]
The Bandwidth Manager dialog lists all USB 3.0 host controllers and GigE network adapters in your computer as well as the cameras connected to each of them. It allows you to allocate a share of the total bandwidth available on the USB 3.0 host controller or network adapter to each camera. This way, you can allocate only so much bandwidth to a camera as it actually needs, thus freeing up resources for other devices.

ı want to control it with python script

christoph-alt-401, 2023-05-04:
Hi! I am interested in the exact same feature as Mehmet. Is there a solution for this?

J700070, 2023-05-25:
You can try using the following code:

### Get the current bandwidth control mode
    current_bandwidth_control_mode = camera.GetNodeMap().GetNode('DeviceLinkThroughputLimitMode')
  
    # Set the bandwidth control mode to 'On'
    current_bandwidth_control_mode.SetValue('On')
  
    # Set the desired bandwidth
    desired_bandwidth = 100000000 
    camera.GetNodeMap().GetNode('DeviceLinkThroughputLimit').SetValue(int(desired_bandwidth))

If you have any further questions, you can check out this link: https://docs.baslerweb.com/device-information-parameters