basler/pypylon #527: EnumerateDevices does not enumerate all connected GigE cameras

Alextraccar, 2022-08-15:
Hi all, as the title suggested, I am calling the following to enumerate two GigE cams I connected to a network switch and then the Mac OS 11.0.1.  But it always only how one cam and it randomly shows one of the two connected.   When I use Pylon Viewer, the two always show up.  I tried to find a solution here but none really worked.  Need your help.  Thanks.

tlf = pylon.TlFactory_GetInstance()
devices = tlf.EnumerateDevices()

if len(devices) == 0:
    raise pylon.RuntimeException("No camera connected")
else:
    print(len(devices))
    for device in devices:
        print(device.GetModelName()) # return readable name
        print(device.GetSerialNumber()) # return unique code

ecmcl, 2022-09-13:
I have the same issue, but working on an arm ubuntu machine.
2 Gige connected cameras, that show up correctly con pylonviewer, but only one camera enumerates with pypylon.

rokitski, 2023-03-04:
Same issue with windows 10 64bit. EnumerateDevices() returns only partial list of cameras on the network. At the same time all cameras show up in Pylon and NI Max explorer. On one computer 7 out of 16 cameras are returned by EnumerateDevices(), while on another machine 9 out of 16 cameras are returned by EnumerateDevices(). Pylon 7.1 is installed on one machine and 7.2 on another. This is basic functionality that needs a solution.

brandonrwin, 2023-05-26:
I have the same problem, M1 Mac, macOS Ventura. I have two cameras, but one shows up twice. If I disconnect one, then the remaining camera shows up once, as it should. I don't think it's python related since `pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(device_info))` results in an error about multiple devices matching.

I tried every combination of Pylon 6.2.0, 6.1.2, and 5.1.1 and all pypylon version tags, with python 3.10. The only combination that works is:

Pylon 5.1.1 with and [pypylon 1.6.0](https://github.com/basler/pypylon/releases/tag/1.6.0), 1.6.0rc1, 1.5.4, or 1.4.0.

All other versions of Pylon resulted in the incorrect device list.

Unmoon, 2024-02-14:
> I have the same problem, M1 Mac, macOS Ventura. I have two cameras, but one shows up twice. If I disconnect one, then the remaining camera shows up once, as it should. I don't think it's python related since `pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(device_info))` results in an error about multiple devices matching.
> 
> I tried every combination of Pylon 6.2.0, 6.1.2, and 5.1.1 (which support macOS) and all pypylon version tags, with python 3.10. The only combination that works is:
> 
> Pylon 5.1.1 with and [pypylon 1.6.0](https://github.com/basler/pypylon/releases/tag/1.6.0), 1.6.0rc1, 1.5.4, or 1.4.0.
> 
> Note, old versions can be added to your requirements.txt/pyproject.toml with "pypylon @ git+https://github.com/basler/pypylon.git@1.6.0".
> 
> All other versions of Pylon resulted in the incorrect device list.

Bless you for this workaround. It seems to still have the issue of devices showing up twice (#152), but at least they all show up.