basler/pypylon #529: Why does saving images to disk affect frame acquisition timing on camera?

aruna-ram, 2022-08-20:
Hi, 

Apologies for the long code! 

- I am filming with two hardware synchronised cameras (model = acA1920-150um). 
- I use the RPi.GPIO library to generate a 100Hz signal, which is received by the cameras
- I start and stop the GPIO (cleanly) when I want to start/stop recording using the script below
- As RAM is not infinite, I save images every 500 frames
- I thought that saving images (a numpy process on the computer) wouldn't affect frames being captured on the camera (a process I thought, not really affected by the computer) 
- I check the difference in timestamps between consecutive frames 
- However it seems that, after one saving process, some frames time differences are very high. An example output:
```
grab loop is 1030
grab loop is 1031
WARNING FRAME SKIPPED: TIME DIFFERENCE BETWEEN FRAMES SHOULD BE 0.01 but is 62.698795929
WARNING FRAME SKIPPED: TIME DIFFERENCE BETWEEN FRAMES SHOULD BE 0.01 but is 62.688869892
grab loop is 1032
grab loop is 1033
```
- this seems to change depending on the frequency of saving 
- From documentation, it seems that timestamp is measured when exposure starts, on the camera (not when it is registered by PC). 
- Why is the saving process on the computer, affecting frame acquisition rate on the camera?

```python
from ast import Raise
import pypylon.pylon as py
import datetime
import matplotlib.pyplot as plt
import numpy as np
import os
#import logging 

NUM_CAMERAS = 2

#file name setup
now = datetime.datetime.now()
dt = now.strftime("%d-%m-%Y:%H:%M")
animal = 'frame_skip_test'
stim_size = '32mm'

#get cameras
tlf = py.TlFactory.GetInstance()
di = py.DeviceInfo()
devs = tlf.EnumerateDevices([di, ])

#create camera array
cam_array = py.InstantCameraArray(NUM_CAMERAS)

#attaches index
for idx, cam in enumerate(cam_array):
    cam.Attach(tlf.CreateDevice(devs[idx]))

cam_array.Open()

# store a unique number for each camera to identify the incoming images
for idx, cam in enumerate(cam_array):
    camera_serial = cam.DeviceInfo.GetSerialNumber()
    print(f"set context {idx} for camera {camera_serial}")
    cam.SetCameraContext(idx)

#set hardware trigger
for camera in cam_array:
    # get clean powerup state
    camera.UserSetSelector = "Default"
    camera.UserSetLoad.Execute()
    camera.ChunkModeActive = True
    camera.LineSelector = "Line4"
    camera.LineMode = "Input"
    camera.TriggerSelector = "FrameStart"
    camera.TriggerMode = "On"
    camera.TriggerSource = "Line4"
    camera.ExposureMode = "Timed"
    camera.ExposureAuto = "Off"
    camera.ExposureTime.SetValue(2000)
    camera.TriggerActivation.SetValue('RisingEdge')
    camera.ChunkSelector = "LineStatusAll"
    camera.ChunkEnable = True
    camera.GainAuto.SetValue('Off')
    camera.Gain.SetValue(9)
    camera.Height.SetValue(1056)
    camera.Width.SetValue(1856)
    if camera.DeviceInfo.GetSerialNumber() == '40088641':
        camera.OffsetX.SetValue(64)
        camera.OffsetY.SetValue(90)
    else:
        camera.OffsetX.SetValue(64)
        camera.OffsetY.SetValue(105)
    print(camera.SensorReadoutTime.GetValue())
    if camera.SensorReadoutTime.GetValue() >= 10000:
        raise Exception("Sensor readout more than 10ms")
    print(camera.ResultingFrameRate.Value)

#height and width
height = cam_array[0].Height.Value
width = cam_array[0].Width.Value
cam_array.MaximumBufferSize = 1000

#make directory to save images ins
#file name setup
now = datetime.datetime.now()
dt = now.strftime("%d-%m-%Y:%H:%M")
dir_name = 'frames_'+dt+animal+stim_size
isExists = os.path.exists(dir_name)
if not isExists:
    os.mkdir(dir_name)
os.chdir(dir_name)
#check directory
pwd = os.getcwd()
print('current directory is {}'.format(pwd))

#containers to store images and timestamps and hardware chunks
time_list_0 = []
time_list_1 = []
io_res_0 = []
io_res_1 = []
img_list_0 = []
img_list_1 = []

#grab loop counter
i = 0
#every n grab loops, save all frames to disk
n = 1000
# file counter
j = 0

cam_array.StartGrabbing()
print('waiting for hardware trigger')
try:
    while True:
        with cam_array[0].RetrieveResult(py.waitForever) as res0:
            with cam_array[1].RetrieveResult(py.waitForever) as res1:
                if res1.GrabSucceeded() and res0.GrabSucceeded():
                    #check correct context
                    img_nr0 = res0.ImageNumber
                    cam_id0 = res0.GetCameraContext()

                    img_nr1 = res1.ImageNumber
                    cam_id1 = res1.GetCameraContext()
                    #print(f"cam #{cam_id}  image #{img_nr}")
                    img0 = res0.GetArray()
                    time_stamp0 = res0.TimeStamp
                    io_values0 = res0.ChunkLineStatusAll.Value
                    res0.Release()

                    img1 = res1.GetArray()
                    time_stamp1 = res1.TimeStamp
                    io_values1 = res1.ChunkLineStatusAll.Value
                    res1.Release()
                    #store images
                    if cam_id0 == 0:
                        img_list_0.append(img0)
                        io_res_0.append((time_stamp0, io_values0))
                    else:
                        print('this shouldn\'t be happening')
                        img_list_1.append(img0)
                        io_res_1.append((time_stamp0, io_values0))
                    if cam_id1 == 1:
                        img_list_1.append(img1)
                        io_res_1.append((time_stamp1, io_values1))
                    else:
                        print('this shouldn\'t be happening')
                        img_list_0.append(img1)
                        io_res_0.append((time_stamp1, io_values1))
                    #check frame skipping 
                    if len(io_res_0)>2:
                        p = len(io_res_0)
                        frame_diff0 = io_res_0[p-1][0] - io_res_0[p-2][0]
                        frame_diff1 = io_res_1[p-1][0] - io_res_1[p-2][0]
                        if frame_diff0/1e9 >= 2/100:
                            print("WARNING FRAME SKIPPED: TIME DIFFERENCE BETWEEN FRAMES SHOULD BE 0.01 but is {}".format(
                                frame_diff0/1.0e9))
                        if frame_diff1/1e9 >= 2/100:
                            print("WARNING FRAME SKIPPED: TIME DIFFERENCE BETWEEN FRAMES SHOULD BE 0.01 but is {}".format(frame_diff1/1e9))
                        print("grab loop is {}".format(i))
                    i += 1
                    #save if n more grab loops executed
                    if i % n == 0:
                        j += 1  # one more file saved
                        print(j)
                        np.save('./multicamera_basicrecord_img0_' +
                                dt + '-{}'.format(j) + '.npy', np.asarray(img_list_0))
                        np.save('./multicamera_basicrecord_img1_' +
                                dt + '-{}'.format(j) + '.npy', np.asarray(img_list_1))
                        np.save('./multicamera_basicrecord_io_res0_' +
                                dt + '-{}'.format(j) + '.npy', np.asarray(io_res_0))
                        np.save('./multicamera_basicrecord_io_res1_' +
                                dt + '-{}'.format(j) + '.npy', np.asarray(io_res_1))

                        #print
                        print('grab loop number = {}'.format(i))
                        print('files saved = {}'.format(j))
                        shape = np.asarray(img_list_0).shape
                        shape1 = np.asarray(img_list_1).shape
                        print('cam 0 array shape is {}'.format(shape))
                        print('cam 1 array shape is {}'.format(shape1))
                        #clear lists
                        img_list_0.clear()
                        img_list_1.clear()
                        io_res_0.clear()
                        io_res_1.clear()
except KeyboardInterrupt:
    #save rest of frames
    j += 1
    np.save('./multicamera_basicrecord_img0_' +
            dt + '-{}'.format(j) + '.npy', np.asarray(img_list_0))
    np.save('./multicamera_basicrecord_img1_' +
            dt + '-{}'.format(j) + '.npy', np.asarray(img_list_1))
    np.save('./multicamera_basicrecord_io_res0_' +
            dt + '-{}-'.format(j) + '.npy', np.asarray(io_res_0))
    np.save('./multicamera_basicrecord_io_res1_' +
            dt + '-{}'.format(j) + '.npy', np.asarray(io_res_1))

    print('grab loop number = {}'.format(i))
    print('TOTAL files saved = {}'.format(j))
    shape = np.asarray(img_list_0).shape
    shape1 = np.asarray(img_list_1).shape
    print('cam 0 array shape is {}'.format(shape))
    print('cam 1 array shape is {}'.format(shape1))

    #clear lists
    img_list_0.clear()
    img_list_1.clear()
    io_res_0.clear()
    io_res_1.clear()
    pass
cam_array.StopGrabbing()
cam_array.Close()
```

SMA2016a, 2022-08-21:
@aruna-ram 

first of all I have a best practice tip: 

Check if you can use camera's user defined name instead of using the SN. This make camera replacement easier later. Hard cording the SN requires the code adaptation when ever you replace the camera.

```python
  if camera.DeviceInfo.GetSerialNumber() == '40088641':
        camera.OffsetX.SetValue(64)
        camera.OffsetY.SetValue(90)
    else:
        camera.OffsetX.SetValue(64)
        camera.OffsetY.SetValue(105)
```

Now to your question:

if you use USB Hub then you need to reduce the camera's Bandwidth also to to 180MB/sec to prevent image losing.

```python
for camera in cam_array:
    # get clean powerup state
    camera.UserSetSelector = "Default"
    camera.UserSetLoad.Execute()
    camera.DeviceLinkThroughputLimit.SetValue(180000000);  # because of USB HUB
    camera.GetStreamGrabberParams().MaxTransferSize.SetValue(4194304);  # for better system performance 
```

if saving an image would  block the grab engine, then the camera will skip images. Please measure the time taken for saving image. Then we can work out a workaround.

aruna-ram, 2022-08-26:
Hi @SMA2016a, thanks for your response and the tips! So it turns out that the issue is running the code on a Mac laptop (not sure why) because I don't get the frame skips when I ran the code on a windows laptop.

closed 2022-08-26 by aruna-ram