basler/pypylon #530: macOS Monterey 12.0.1 pypylon installation M1

rphulwani, 2022-08-22:
Hello,

When I try to install pypylon and import it to use it, I get this error:

Traceback (most recent call last):
  File "/Users/.../Documents/Code/.../try.py", line 1, in <module>
    from pypylon import pylon
  File "/Users/.../Documents/Code/.../newvenv/lib/python3.10/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: dlopen(/Users/.../Documents/Code/.../newvenv/lib/python3.10/site-packages/pypylon/_pylon.cpython-310-darwin.so, 0x0002): tried: '/Users/.../Documents/Code/.../newvenv/lib/python3.10/site-packages/pypylon/_pylon.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/local/lib/_pylon.cpython-310-darwin.so' (no such file), '/usr/lib/_pylon.cpython-310-darwin.so' (no such file)

I've tried changing arch to x86_64 and reinstalling using pip but I get the same error.

I used the pylon 6.2.0 package installer for MacOS since that seems like the latest version for Mac

I have a virtual env and used
```pip install pypylon```

thiesmoeller, 2022-08-22:
For now Pylon and pypylon are only available for x86_64.

So have to run from emulated environment on you M1 mac

rphulwani, 2022-08-22:
Do you mean something to do with arch or Rosetta? Not sure what you mean by emulated environment

thiesmoeller, 2022-08-22:
Yes I mean rosetta

rphulwani, 2022-08-23:
Alright I got it thanks to this answer: [https://stackoverflow.com/a/64951025](url)

- first install separate version of homebrew to /usr/local/ 
```cd /usr/local```
- install Intel x86_64 emulated homebrew
```arch --x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"```
- create alias for Intel x86_64 emulated homebrew
```nano ~/.zshrc```
- add this line to ~/.zshrc file
```alias brow='arch --x86_64 /usr/local/Homebrew/bin/brew'```
- Confirm
```which brow``` should return ```brow: aliased to arch --x86_64 /usr/local/Homebrew/bin/brew```
```brow --prefix``` should return ```/usr/local```
- Install python3.10 using intel homebrew
```brow install python@3.10```
- Start editing ~/.zshrc file
```nano ~/.zshrc```
- Add intel python3.10 to path by adding following line to ~/.zshrc file
```export PATH="/usr/local/opt/python@3.10/bin:$PATH"```
- Open new terminal
- Confirm
```which python3``` should return ```/usr/local/opt/python@3.10/bin/python3```
- Install pypylon
```pip3 install pypylon```
- Use library

closed 2022-08-23 by rphulwani

kopecn, 2023-04-25:
Hey, when should we expect a fix?  Our business unit is starting to switch over to Ximea's which provides native support.

m-binev, 2023-05-05:
@kopecn Hi, we hope to provide a native support next year. Would you like to connect with me offline to elaborate on your requirements so that I could better nail the business case and give it more priority at Basler? Thanks & regards, Momchil