basler/pypylon #531: How to acquire at high fps?

Lucrezia-Cester, 2022-08-23:
Hello, 

I am trying to acquire images at 500 fps. When using the Pylon Software I managed to do this by simply reducing the number of pixels that are recording and increasing fps. 

How can I do the same thing with python? 

I am using the grab.py example.

Can I just add for the fps sopmething like  cam.fps = 1000 Hz?

And what about the code to reduce the number of acquiring pixels, so that only a small subset of pixels acquires data?

Lucrezia-Cester, 2022-08-24:
UPDATE: 

just used the features Height and Width to crop ROI. 

One more question tho, how can I check fps?

thiesmoeller, 2022-08-24:
https://docs.baslerweb.com/resulting-frame-rate#checking-the-resulting-frame-rate

closed 2022-12-06 by Lucrezia-Cester