basler/pypylon #532: Action command not working

shrinand1996, 2022-08-25:
I have 2 basler ace cameras connected to a POE switch. I setup action commands and PTP in pylon viewer and they worked perfectly fine. But when i use pypylon to set Action Commands, it crashes with the error :
using acA4024-8gc @ 192.168.0.4
ActionGroupKey 0x0
ActionGroupMask 0x0
TriggerSource Line1
TriggerMode Off
AcquisitionMode SingleFrame

ActionGroupKey 0x112233
ActionGroupMask 0xffffffff
TriggerSource Action1
TriggerMode On
AcquisitionMode Continuous

issuing no-wait action command
Traceback (most recent call last):
  File "test.py", line 101, in <module>
    with cam.RetrieveResult(1000, pylon.TimeoutHandling_ThrowException):
  File "/home/jetson/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 3502, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1037)

Please help. When i increase the timeout in retrieveResult, it grabs 1 frame and then throws this error.

SMA2016a, 2022-08-27:
In which frequency do you issue the action command?   
Would you post minimum code that can replicate the effect?

SMA2016a, 2024-03-31:
Since we have wait long enough, I close the case.

closed 2024-03-31 by SMA2016a