basler/pypylon #533: How to get the node type?

luckygg, 2022-09-07:
Hi all.
I have a question about the GenICam node type.

I would like to show all the node list of the camera.
I could access all camera's nodes but couldn't get the type of each node.

For example, 'Width' is Integer type, 'TriggerSelector' is Enumeration type.

Could you please let me know how to get each node type?

Thank you!

thiesmoeller, 2022-09-08:
As a quick python 3.10+ example see this code:

```python
import pypylon.pylon as py
import pypylon.genicam as geni
import queue

tlf = py.TlFactory.GetInstance()

cam = py.InstantCamera(tlf.CreateFirstDevice())

cam.Open()

worklist = queue.LifoQueue()

worklist.put(cam.Root)

while not worklist.empty():
    n = worklist.get()
    print(n.Node.Name, end=":\t")
    match type(n):
        case geni.IBoolean:
            print("Bool.")
        case geni.ICommand:
            print("Command")            
        case geni.IInteger:
            print("Integer")
        case geni.IEnumeration:
            print("Enumeration")
        case geni.IFloat:
            print("Float")
        case geni.IString:
            print("String")
        case geni.IRegister:
            print("Register")
        case geni.ICategory:
            print("Category ----->")
            # push children of category
            children = n.Node.GetChildren()
            for c in children:
                worklist.put(c)
        case _:
            print(type(n))
cam.Close()

```

luckygg, 2022-09-13:
@thiesmoeller Thank you for your reply.
It is so simple!

closed 2022-09-13 by luckygg