basler/pypylon #535: Synchronous Multi-Camera System; Bandwidth Problem

AIGR-sw, 2022-09-24:
For a computer vision application, I would like to synchronously operate four GigE cameras (a2A2448-23gcBAS) that are connected via an Ethernet switch to my PC. For this purpose, I use _Precision Time Protocol (PTP)_ together with the _Periodic Signal_ feature to capture images synchronously on all cameras at regular time intervals. Further I configured the cameras and the Ethernet adapter as follows: 

- Network Configuration (GigE Cameras) according to the description on the docs.baslerweb.com website (Jumbo Frames enabled, MTU=9000)
- All network components comply with Basler recommendations (were recommended by Basler sales consulting). (Switch: Netgear MS510TXPP; Cables: GigE cabel Cat.6 Art. Nr: 1003708; Ethernet Adapter: Intel I219-LM, 1Gbit/s)
- I make use of the _Pixel Beyond_ feature (scaling factor 2.8) and _Compression Beyond_ feature (compression ratio 70%) to reduce the required bandwidth
- I used Basler's Bandwidth Manager to determine the _Inter-Packet Delay_ and _Packet Size_ settings in the Transport Layer category

Despite these settings and configurations, I receive the following error message when running the Python script attached:

> Error:  3774873620 The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size. 

Can you tell me what I need to change / optimize / consider to fix this error?

```
framerate = 20.
exposure = 20.
gain = 1.
compression_ratio = 70.
pixel_scaling = 2.79

def init_ptp(camera, index):
    # PTP setup
    camera.PtpEnable.SetValue(False)
    camera.BslPtpPriority1.SetValue(128)
    camera.BslPtpProfile.SetValue("DelayRequestResponseDefaultProfile")
    camera.BslPtpNetworkMode.SetValue("Multicast")
    camera.BslPtpManagementEnable.SetValue(False)
    camera.BslPtpTwoStep.SetValue(False)
    camera.PtpEnable.SetValue(True)

def check_ptp(cameras):
    initialized = False
    locked = False

    # Wait until correctly initialized or timeout
    time1 = time.time()
    while not initialized:
        status_arr = np.zeros(cameras.GetSize(), dtype=np.bool8)
        for i, camera in enumerate(cameras):
            camera.PtpDataSetLatch.Execute()
            status_arr[i] = (camera.PtpStatus.GetValue() == 'Master' \
                            or camera.PtpStatus.GetValue() != 'Initializing')
        initialized = np.all(status_arr)
        if (time.time() - time1) > 3:
            if not initialized:
                warnings.warn('PTP not initialized -> Timeout')
            break

    # If correctly initialized, wait until settled or timeout
    if initialized:
        time2 = time.time()
        while not locked:
            status_arr = np.zeros(cameras.GetSize(), dtype=np.bool8)
            status_string = ''
            for i, camera in enumerate(cameras):
                camera.PtpDataSetLatch.Execute()
                status_arr[i] = (camera.PtpStatus.GetValue() == 'Master' \
                                or camera.PtpServoStatus.GetValue() == 'Locked')
                status_string += 'Camera {:d} locked: {} | '.format(i, status_arr[i])
            print(status_string)
            locked = np.all(status_arr)
            if (time.time() - time2) > 30:
                if not locked:
                    warnings.warn('PTP not locked -> Timeout')
                break

    return initialized and locked

def init_camera(camera, index):
    camera.GainAuto.SetValue("Off")
    camera.Gain.SetValue(gain)

    camera.ExposureAuto.SetValue("Off")
    camera.ExposureTime.SetValue(int(exposure*1000))

    # Beyond Pixel setup
    camera.PixelFormat.SetValue("RGB8")
    camera.BslScalingFactor.SetValue(pixel_scaling)

    # Beyond Compression setup
    camera.ImageCompressionMode.SetValue("BaslerCompressionBeyond")
    camera.ImageCompressionRateOption.SetValue("FixRatio")
    camera.BslImageCompressionRatio.SetValue(compression_ratio)

    # Periodic Signal setup
    if camera.BslPeriodicSignalSource.GetValue() != 'PtpClock':
        warnings.warn('Clock source of periodic signal is not `PtpClock`')
    camera.BslPeriodicSignalPeriod.SetValue(1 / framerate * 1e6)
    camera.BslPeriodicSignalDelay.SetValue(0)
    camera.TriggerSelector.SetValue("FrameStart")
    camera.TriggerMode.SetValue("On")
    camera.TriggerSource.SetValue("PeriodicSignal1")

    # Transport Layer Control
    camera.GevSCPD.SetValue(222768)
    camera.GevSCFTD.SetValue(8018*index)
    camera.GevSCPSPacketSize.SetValue(8000)

# Get the transport layer factory
tlFactory = pylon.TlFactory.GetInstance()

# Get all attached devices and exit application if no device is found
devices = tlFactory.EnumerateDevices()
cam_count = len(devices)
if not cam_count:
    raise EnvironmentError('No camera device found')

# Create and attach all Pylon Devices
cameras = pylon.InstantCameraArray(cam_count)
for camera, device in zip(cameras, devices):
    print('Using {:s} @ {:s}'.format(device.GetModelName(), device.GetIpAddress()))
    camera.Attach(tlFactory.CreateDevice(device))
    camera.Open()

# Initialize PTP and check initialization
for i, camera in enumerate(cameras):
    init_ptp(camera, i)

success = check_ptp(cameras)
if not success:
    raise EnvironmentError('PTP initialization was not successful')

# Initialize general camera parameters
for i, camera in enumerate(cameras):
    init_camera(camera, i)

# Image decompression
decompressor = pylon.ImageDecompressor()
descriptor = cameras[0].BslImageCompressionBCBDescriptor.GetAll()
decompressor.SetCompressionDescriptor(descriptor)

# Prepare image grabbing
imgs = [None] * cam_count
ids = [None] * cam_count

cv2.namedWindow('image', cv2.WINDOW_NORMAL)

cameras.StartGrabbing(pylon.GrabStrategy_LatestImages)

while True:
    with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
        idx = grabResult.GetCameraContext()

        if grabResult.GrabSucceeded():
            payload = grabResult.GetBuffer()
            decompressed_image = decompressor.DecompressImage(payload)

            imgs[idx] = decompressed_image.Array
            ids[idx] = grabResult.ID

            # Display image
            if ids.count(ids[0]) == len(ids):
                cv2.imshow('image', np.concatenate(imgs, axis=1)[...,::-1]/255.0)
                if cv2.waitKey(5)==27:
                    break

        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)

cameras.StopGrabbing()
cameras.Close()
```

SMA2016a, 2022-09-24:
You need to do the image processing outside the the grab loop or you need to run the grab loop in an individual thread.  

Please the time that consume this block on your PC.   The time should not be higher than 1/fps.

```python
if grabResult.GrabSucceeded():
            payload = grabResult.GetBuffer()
            decompressed_image = decompressor.DecompressImage(payload)

            imgs[idx] = decompressed_image.Array
            ids[idx] = grabResult.ID

            # Display image
            if ids.count(ids[0]) == len(ids):
                cv2.imshow('image', np.concatenate(imgs, axis=1)[...,::-1]/255.0)
                if cv2.waitKey(5)==27:
                    break

````

AIGR-sw, 2023-02-01:
@SMA2016a you're right. The loop period takes longer than 1/fps.

Unfortunately, the decompression of the image takes more than 8 ms per image, and I have to decompress four images. Is there any way to speed up the decompression of the images? E.g. decompressing four images at once or a faster alternative to Pypylon's ImageDecompressor.

thiesmoeller, 2023-02-01:
Your PTP trigger runs at 20hz
So you have 50ms to handle the four frames. If decompression takes ~32ms ... The pure capturing should be fast enough.

I would expect the concatenation and display to be the problematic part.

Have you timed these parts also?

AIGR-sw, 2023-02-02:
Hey @thiesmoeller, thanks for your answer!
My trigger signal is currently running at 25Hz and the Python code looks like this:

```python
while True:
    with self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
        if grabResult.GrabSucceeded():
            decompressed_image = self.decompressor.DecompressImage(grabResult.Buffer)
            decompressed_array = decompressed_image.Array
            decompressed_image.Release()
            cam_idx = grabResult.GetCameraContext()
            imgs = frame_buffer.append(grabResult.TimeStamp*1e-9, cam_idx, decompressed_array)

            # If set of images is available
            if imgs is not None:
                img_resized = self._resize_images(imgs)
                self.output_queue.put(img_resized)
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
```
Tracing the code with Viztracer shows that decompression takes most of the time:
[image]

Depending on the CPU load caused by the image processing task (which is executed in another Python process), decompressing the images may take even longer than a loop iteration, which causes this problem.

I would appreciate any help to speed up the decompression. Maybe there exists a way to decompress all four images at once?

thiesmoeller, 2023-02-07:
While fixing a memory leak on decompress image, I removed some unnecessary copies... which should gain some ms...

Can you check the pre-release https://pypi.org/project/pypylon/1.9.0rc2/ ?

Next step on our side is to check multithreading while decompression. Current implementation blocks the GIL during decompression. So you would have to use multiprocessing to run in parallel.

If you want to put the images side by side I recommend to use slice notation, which will prevent a number of memcpy operations:

```python
width = cameras[0].Width.GetValue()
height = cameras[0].Height.GetValue()

# I tested with MONO8 ... you would have to add the color channel as third dimension
imgs = np.zeros((height, width*cam_count), dtype=np.uint8)
....
cameras.StartGrabbing(100, pylon.GrabStrategy_LatestImages)
while True:
    with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
        idx = grabResult.GetCameraContext()

        if grabResult.GrabSucceeded():
            # decompress image can handle the grabresult directly ... skipping one memcpy
            decompressed_image = decompressor.DecompressImage(grabResult)
            # use slice notation to enter array
            imgs[:,width*idx:width*(idx+1)] = decompressed_image.GetArray()

```

closed 2023-04-06 by AIGR-sw