basler/pypylon #536: how to change trigger mode from hardware to software ?

ridvanozdemir, 2022-09-26:
Hello,

I want to start capturing an image with a hardware trigger and continue capturing with software trigger? How can I do that?

ridvanozdemir, 2022-09-28:
from pypylon import pylon
from pypylon import genicam
import sys
import cv2
import time
import numpy as np

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.TriggerMode ='On'
camera.TriggerSource ='Line1'
camera.TriggerSelector ='FrameStart'
camera.TriggerActivation='RisingEdge'

print("Trigger mode: ", camera.TriggerMode.Value)
camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

i=0
while camera.IsGrabbing():
    print("tetik ver")    
    if camera.TriggerSource.Value == 'Line1' and i ==1:
        camera.TriggerSource ='Software'        
    if camera.TriggerSource.Value == 'Software':
        camera.TriggerSoftware.Execute()        
    grabResult = camera.RetrieveResult(15000, pylon.TimeoutHandling_ThrowException)
    
    if grabResult.GrabSucceeded():
        print(str(i), ". görüntü alindi")
        image = converter.Convert(grabResult)
        image = image.GetArray()
        file = "image" + str(i) 
        image = image[(grabResult.GetHeight()//2)-300:(grabResult.GetHeight()//2)+300, (grabResult.GetWidth()//2)-300:(grabResult.GetWidth()//2)+300]
        globals()['img%s' % i] = image
        filename= file + ".jpg"
        cv2.imwrite(filename, image)
        time.sleep(1)
        i+=1
        if i==3:
            vis = np.concatenate((img0, img1, img2), axis=1)
            cv2.imwrite("final_image.jpg", vis)
            grabResult.Release()
            # Releasing the resource    
            camera.StopGrabbing()
            camera.Close()
            break

ridvanozdemir, 2022-09-28:
I made it but it always get the same image? 
where should I fix?

thiesmoeller, 2022-09-28:
some inline comments and only softwaretrigger for quick test
-> this works
Side note: for higher quality results and less CPU load on your system I really recommend to simply use the YUV ( then with cv2.cvtColor ) or the RGB output ( if camera supports it)

```python
from pypylon import pylon
import cv2
import time
import numpy as np

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
# Always Selector first!
camera.TriggerSelector = 'FrameStart'
camera.TriggerMode = 'On'
camera.TriggerSource = 'Software'
camera.TriggerActivation = 'RisingEdge'
# for test moving testimage
camera.TestImageSelector = "Testimage4"

print("Trigger mode: ", camera.TriggerMode.Value)
camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

def crop_center_square(img, size):
    h, w = img.shape[:2]
    return img[h // 2 - size // 2:h // 2 + size // 2, w // 2 - size // 2:w // 2 + size // 2:]

img = []
i = 0
while camera.IsGrabbing():
    print("tetik ver")
    # only use sw trigger
    # if camera.TriggerSource.Value == 'Line1' and i == 1:
    #    camera.TriggerSource = 'Software'
    if camera.TriggerSource.Value == 'Software':
        camera.TriggerSoftware.Execute()

    grabResult = camera.RetrieveResult(15000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        print(i, ". görüntü alindi")
        image = converter.Convert(grabResult)
        # release early
        grabResult.Release()

        image = image.GetArray()
        file = f"image{i}"

        image = crop_center_square(image, 600)
        img.append(image)
        filename = file + ".jpg"
        cv2.imwrite(filename, image)
        time.sleep(1)
        i += 1
        if i == 3:
            vis = np.concatenate(img, axis=1)
            cv2.imwrite("final_image.jpg", vis)
            break

camera.StopGrabbing()
camera.Close()
```

ridvanozdemir, 2022-09-28:
thanks for quick response. 
Is it possible to change TriggerSource while grabbing? or should I stop grabbing then change the  TriggerSource from Line1 to Software then start grabbing again?

I have tried some options but it didn't work out.

thiesmoeller, 2022-09-28:
Switching during runtime should work....
But:
What do you want to get eventually from your program ?

It looks like:
For one hardware trigger get three images 
If this is the case, we have a special feature on a lot of cameras for this:

https://docs.baslerweb.com/trigger-selector#frame-burst-start-acquisition-start-end-and-active-trigger

It is burst acquisition. You configure the number of images you want to get. The sequence starts with the trigger and the following frames are automatically started with the configured frame rate.
Main benefit is that you remove all the jitter introduced by the software trigger

ridvanozdemir, 2022-09-29:
yes, Frame Burst Start can solve my problem. I just need to start capturing image with a trigger after that I need to capture 2 more images with 200ms delay.

I couldn't find any example of frame burst start, can you give me a sample code which shows how to save 3 different image with Frame Burst?

ridvanozdemir, 2022-09-30:
Thank you Mr. Möller, that program really works very good.
I have one step left. 

when I use while loop I can't update my PyQt5 user interface but it works fine with ImageEventHandler.

Can I implement "FrameBurstStart" to 
camera.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)?

thiesmoeller, 2022-09-30:
The way you capture on your host ( loop or callback handler ) is independent from the trigger setup in the camera.

So yes this will work in your pyqt scenario.

ridvanozdemir, 2022-09-30:
I have tried to use GrabLoop with FrameBurstStart but it didn't work.

here is my code.

import pypylon.pylon as py
import numpy as np
import cv2

tlf = py.TlFactory.GetInstance()
camera = py.InstantCamera(tlf.CreateFirstDevice())
camera.Open()
camera.RegisterImageEventHandler(ImageEventPrinter(), py.RegistrationMode_Append, py.Cleanup_Delete)

# disable FrameStart trigger
camera.TriggerSelector = "FrameStart"
camera.TriggerMode = "Off"

# configure FrameBurstStart trigger
camera.TriggerSelector = "FrameBurstStart"
camera.TriggerMode = "On"
# for testing via software
#cam.TriggerSource = "Software"
camera.TriggerSource = "Line1"
camera.TriggerActivation = "RisingEdge"
camera.AcquisitionBurstFrameCount = 3

# configure a 200ms spacing between images:
frameperiod_s = 200e-3
camera.AcquisitionFrameRateEnable = True
camera.AcquisitionFrameRate = 1 / frameperiod_s

camera.StartGrabbing(py.GrabStrategy_LatestImageOnly, py.GrabLoop_ProvidedByInstantCamera)

grab_times = []

converter = py.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = py.PixelType_BGR8packed
converter.OutputBitAlignment = py.OutputBitAlignment_MsbAligned

global i
i=0
class ImageEventPrinter(py.ImageEventHandler):
    def OnImagesSkipped(self, camera, countOfSkippedImages):
        print("OnImagesSkipped event for device ", camera.GetDeviceInfo().GetModelName())
        print(countOfSkippedImages, " images have been skipped.")

    def OnImageGrabbed(self, camera, grabResult):
        print("OnImageGrabbed event for device ", camera.GetDeviceInfo().GetModelName())
        print('Waiting for trigger')
        if grabResult.GrabSucceeded():
            grab_times.append(grabResult.TimeStamp)
            img = grabResult.Array
            # do your storage
            image = converter.Convert(grabResult)
            image = image.GetArray()
            print(str(i), ". image captured")
            file = "image" + str(i) 
            image = image[(grabResult.GetHeight()//2)-300:(grabResult.GetHeight()//2)+300, (grabResult.GetWidth()//2)-300:(grabResult.GetWidth()//2)+300]
            globals()['img%s' % i] = image
            filename= file + ".jpg"
            cv2.imwrite(filename, image)
            i+=1
            # do your storage
        if len(grab_times) == 3:
            # stop after three frames
            vis = np.concatenate((img0, img1, img2), axis=1)
            cv2.imwrite("final_image.jpg", vis)
            print("program terminated")
            print("LineStatus.Value", camera.LineStatus.Value)
            
# show frameperiod
print(f"acquired {len(grab_times)} frames")
delta_times = np.diff(grab_times)
print(f"time between frames is {delta_times / 1e6}ms")

and the output is:

acquired 0 frames
time between frames is []ms

thiesmoeller, 2022-09-30:
Can you please reformat your code with proper markdown first

````
```python
.... Your code ....
```
````

ridvanozdemir, 2022-09-30:
Here is my code.

```python
import pypylon.pylon as py
import numpy as np
import cv2

tlf = py.TlFactory.GetInstance()
camera = py.InstantCamera(tlf.CreateFirstDevice())
camera.Open()
camera.RegisterImageEventHandler(ImageEventPrinter(), py.RegistrationMode_Append, py.Cleanup_Delete)

disable FrameStart trigger
camera.TriggerSelector = "FrameStart"
camera.TriggerMode = "Off"

configure FrameBurstStart trigger
camera.TriggerSelector = "FrameBurstStart"
camera.TriggerMode = "On"

for testing via software
#cam.TriggerSource = "Software"
camera.TriggerSource = "Line1"
camera.TriggerActivation = "RisingEdge"
camera.AcquisitionBurstFrameCount = 3

configure a 200ms spacing between images:
frameperiod_s = 200e-3
camera.AcquisitionFrameRateEnable = True
camera.AcquisitionFrameRate = 1 / frameperiod_s

camera.StartGrabbing(py.GrabStrategy_LatestImageOnly, py.GrabLoop_ProvidedByInstantCamera)

grab_times = []

converter = py.ImageFormatConverter()

converting to opencv bgr format
converter.OutputPixelFormat = py.PixelType_BGR8packed
converter.OutputBitAlignment = py.OutputBitAlignment_MsbAligned

global i
i=0
class ImageEventPrinter(py.ImageEventHandler):
def OnImagesSkipped(self, camera, countOfSkippedImages):
print("OnImagesSkipped event for device ", camera.GetDeviceInfo().GetModelName())
print(countOfSkippedImages, " images have been skipped.")

def OnImageGrabbed(self, camera, grabResult):
    print("OnImageGrabbed event for device ", camera.GetDeviceInfo().GetModelName())
    print('Waiting for trigger')
    if grabResult.GrabSucceeded():
        grab_times.append(grabResult.TimeStamp)
        img = grabResult.Array
        # do your storage
        image = converter.Convert(grabResult)
        image = image.GetArray()
        print(str(i), ". image captured")
        file = "image" + str(i) 
        image = image[(grabResult.GetHeight()//2)-300:(grabResult.GetHeight()//2)+300, (grabResult.GetWidth()//2)-300:(grabResult.GetWidth()//2)+300]
        globals()['img%s' % i] = image
        filename= file + ".jpg"
        cv2.imwrite(filename, image)
        i+=1
        # do your storage
    if len(grab_times) == 3:
        # stop after three frames
        vis = np.concatenate((img0, img1, img2), axis=1)
        cv2.imwrite("final_image.jpg", vis)
        print("program terminated")
        print("LineStatus.Value", camera.LineStatus.Value)
show frameperiod
print(f"acquired {len(grab_times)} frames")
delta_times = np.diff(grab_times)
print(f"time between frames is {delta_times / 1e6}ms")
.... 
```

thiesmoeller, 2022-10-01:
This is non functional python code.

ridvanozdemir, 2022-10-03:
this is the whole code for my project, I have setPhoto function but it doesn't update the image in PyQt5 GUI label when I use grab_kram function.

```python
from PyQt5.QtWidgets import QMainWindow, QApplication, QPushButton, QLabel, QFileDialog
from PyQt5.QtGui import *
from PyQt5.QtCore import *
from PyQt5 import uic
import sys
import cv2
import imutils
import numpy as np
import os
import glob
import shutil
import snap7
import struct
from pypylon import pylon
import argparse
import os
import glob
import random
import time
import numpy as np
from pypylon import genicam
import platform
import datetime

total_time = time.time()

"""

plc = snap7.client.Client()
#plc.connect('10.10.8.206', 0, 1)  # IP address, rack, slot (from HW settings)
plc.connect('192.168.0.1', 0, 1)  # IP address, rack, slot (from HW settings)

#PLC
db_number = 2
start_offset = 2
bit_offset = 1
value = 0  # 1 = true | 0 = false

start_address = 100  # starting address
length = 4  # double word

def writeBool(db_number, start_offset, bit_offset, value):
	reading = plc.db_read(db_number, start_offset, 1)    # (db number, start offset, read 1 byte)
	snap7.util.set_bool(reading, 0, bit_offset, value)   # (value 1= true;0=false) (bytearray_: bytearray, byte_index: int, bool_index: int, value: bool)
	plc.db_write(db_number, start_offset, reading)       #  write back the bytearray and now the boolean value is changed in the PLC.
	return None

def readBool(db_number, start_offset, bit_offset):
	reading = plc.db_read(db_number, start_offset, 1)  
	a = snap7.util.get_bool(reading, 0, bit_offset)
	print('DB Number: ' + str(db_number) + ' Bit: ' + str(start_offset) + '.' + str(bit_offset) + ' Value: ' + str(a))
	return None

def readMemory(start_address,length):
	reading = plc.read_area(snap7.types.Areas.MK, 0, start_address, length)
	value = struct.unpack('>f', reading)  # big-endian
	print('Start Address: ' + str(start_address) + ' Value: ' + str(value))

def writeMemory(start_address,length,value):
	plc.mb_write(start_address, length, bytearray(struct.pack('>f', value)))  # big-endian
	print('Start Address: ' + str(start_address) + ' Value: ' + str(value))

writeBool(db_number, start_offset, bit_offset, value)
#
readBool(db_number, start_offset, bit_offset)

writeMemory(start_address, length, 26.78)
[... 284 lines cut ...]
                        vis = np.concatenate((img0, img1, img2), axis=1)
                        cv2.imwrite("final_image.jpg", vis)
                        self.setPhoto(vis)
                        #self.setPhoto(vis)
                        print("program terminated")
                        grabResult.Release()
                        camera.TriggerSource ='Line1'
                        #break
                        i=0
            except Exception as e:
                print(e)
                camera.StopGrabbing()
                camera.Close()
                break
        #self.setPhoto(vis)
        return vis           
       
app = QApplication(sys.argv)
UIWindow = UI()
#MainWindow = QMainWindow()
#ui = Gui.Ui_MainWindow()
#ui.setupUi(MainWindow)
UIWindow.show()
app.exec_()   
```

closed 2022-10-11 by ridvanozdemir