basler/pypylon #538: How to change the field of view of the camera ?

Lucrezia-Cester, 2022-09-27:
I am setting values for camera.height and .width, but I would like to personally choose which area it is looking at, for example, if I only wanted to capture from the 10x10 pixels from the top right corner vs left right corner, how would I go about that>

Lucrezia-Cester, 2022-09-27:
I am trying to follow this instructions https://docs.baslerweb.com/image-roi but it doesn't seem to shift the FOV

thiesmoeller, 2022-09-28:
Sidenote: Not all camera models can create an image as small as a  10x10 image. There might be limits like 64x64

But in principle the flow is
For a top left 10x10 image:

```
cam.Width = 10
cam.Height =10
cam.OffsetX = 10
cam.OffsetY = 10
```

And to move this to top right
```
cam.OffsetX = cam.OffsetX.Max
```

closed 2024-04-08 by SMA2016a