basler/pypylon #540: Getting inconsistent timed pictures by hardware triggering

stewie6514, 2022-10-07:
Hello,‎

I'm trying to get a camera triggered by an external signal. My set up  consists of a a2a2590-60umbas and a  raspberry pi 4 with Raspberry Pi OS and ‎the pylyon viewer installed. The trigger is an analog sinusoidal signal that I can adjust for ‎offset and gain. I set it to get a sine wave between 0-3V with a frequency of 1hz.  I feed this ‎signal to one of the two GPIO lines. This gives me a fairly consistent trigger signal to begin ‎with. A picture of the signal arriving at my camera is attached. Based on this signal I try to ‎take pictures with the camera, which triggers only in irregular time intervals. The code is ‎based on the example "USB_hardware_trigger_and_chunks.ipynb " from @thiesmoeller and ‎I made only slight changes:‎
```python
import pypylon.pylon as py
import pypylon.genicam as geni
import matplotlib.pyplot as plt
import numpy as np

# open the camera
tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

‎#get clean powerup state‎
cam.UserSetSelector = "Default"‎
cam.UserSetLoad.Execute()‎

# set the camera options to max speed
cam.PixelFormat.SetValue('Mono8')
cam.Height = cam.Height.Min
cam.Width = cam.Width.Min
cam.ExposureTime = cam.ExposureTime.Min
# disable exposure auto
cam.ExposureAuto.SetValue('Off')
#Set Gain to zero and turn off exposure auto
cam.GainAuto.SetValue('Off')
cam.Gain.SetValue(0)

‎#setup the io section‎
cam.LineSelector = "Line2"‎
cam.LineMode = "Input"‎

‎#setup the trigger / acquisition controls‎
cam.TriggerSelector = "FrameStart"‎
cam.TriggerSource = "Line2"‎
cam.TriggerMode = "On"‎
camera.TriggerDelay.SetValue(0.0)
camera.BslInputHoldOffTime.SetValue(20000)

# Set the trigger activation mode to continuous and rising edge
cam.AcquisitionMode.SetValue('Continuous')
cam.TriggerActivation.SetValue('RisingEdge')

res = cam.GrabOne(py.waitForever)‎

‎#definition of event handler class ‎
class TriggeredImage(py.ImageEventHandler):‎
‎    def __init__(self):‎
‎        super().__init__()‎
‎        self.grab_times = []‎
‎    def OnImageGrabbed(self, camera, grabResult):‎
‎        self.grab_times.append(grabResult.TimeStamp)‎
‎#create event handler instance‎
image_timestamps = TriggeredImage()‎

‎#register handler‎
‎#remove all other handlers‎
cam.RegisterImageEventHandler(image_timestamps, ‎
‎                              py.RegistrationMode_ReplaceAll, ‎
‎                              py.Cleanup_None)‎

‎#start grabbing with background loop‎
cam.StartGrabbingMax(20, py.GrabStrategy_LatestImages, ‎py.GrabLoop_ProvidedByInstantCamera)‎
‎#wait ... or do something relevant‎
while cam.IsGrabbing():‎
‎    time.sleep(0.1)‎
‎#stop grabbing‎
cam.StopGrabbing()‎

frame_delta = np.diff(np.array(image_timestamps.grab_times))/1e6
print(frame_delta)

cam.Close()
```

This leads to Time differences between the pictures of:‎
‎[ 19.16996 447.11774  19.19745 520.02957  19.16995 441.8656   19.26782‎
‎ 518.78563  19.16995 442.87258  19.16995 518.88397  19.16995 443.54366‎
‎  19.16995 513.39529  19.16995  78.06384 368.22616]‎

Why are the images taken at such different time intervalls?‎
[image]

thiesmoeller, 2022-10-07:
With a sine wave to trigger you're not following the recommendations at https://docs.baslerweb.com/io-timing-characteristics-(ace-ace-2-boost)#recommendations.

Overall:
The steeper your input edges are, the less trouble...

To go into your use case:
Your frameperiod ( 1/FPS ) is pretty short due to all relevant features set to minimum.
So the camera can be retriggered really fast 

Your signal is 1Hz your holdoff time is 20ms

The trigger timestamps you showed match exactly with a drawing of a sinewave in your setup:

You trigger multiple times ( with 20ms spacing ) in the problematic area at the front and back of the wave.
Because here slight fluctuations in the signal are detected as rising edge -> and a trigger gets executed.

Simple solution is to extend the hold off time to nearly one second. 
Than the first trigger on the rising edge will trigger your frame ... and any further noise up to the next 1hz "pulse" gets ignored.

the green lines in the drawing are exactly your numbers:
[image]

stewie6514, 2022-10-14:
Thank you for the fast and very well explained answer. Indeed, the camera triggered twice ‎during the upper part of the sine wave. I used a timer to extended the holdoff time to about ‎‎0,5s and got very consistent timings.

thiesmoeller, 2022-10-14:
that means you can close the issue?

closed 2022-10-26 by thiesmoeller