basler/pypylon #542: About GrabOne processing time

KazukiYumiba, 2022-10-11:
I got pypylon/samples/grabone.py working.
At that time, the processing of line 17: GrabOne () speed was measured using the time module.

`start =time.time()
result = camera. GrabOne(100)
print(time.time() - s)`

The camera used is acA1600-60gm.
The measurement time was about **120ms~140ms.**
Is there a way to speed up the above process?
If not, I would like to know alternatives.

By the way, when I measured line 51 of pypylon/samples/grab.py , it was **16ms to 17ms** and it behaved as expected.

`start = time.time()
grabResult = camera.RetrieveResult(5000,pylon.TimeoutHandling_ThrowException)
print(time.time() - s)`

please help me! !

thiesmoeller, 2022-10-11:
With GrabOne you start and stop the complete grab engine for only a single frame. This is not efficient, if you want to capture more than one frame.

Typical pattern is normally
StartGrabbing -> start capture engine

Capture frames 

StopGrabbing()

See some more examples here: https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

KazukiYumiba, 2022-10-17:
Thanks for sending me the sample code!
I understood.

closed 2022-10-17 by KazukiYumiba