basler/pypylon #544: _genicam.RuntimeException: NULL pointer dereferenced : RuntimeException thrown (

Simonm952, 2022-10-13:
Hi,

I am using a Nvidia Xavier NX device with JetPack 4.6, L4T 32.6.1 OS flashed on an SSD, so it is getting booted from this external storage. I connect an USB Basler camera to this device.

I am using pylon version 6, pypylon 1.8.0 and python 3.7.

When running 

`pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice()) `

I am getting an error: 
 
`NULL pointer dereferenced : RuntimeException thrown (file 'PylonPowerNodeMap.h', line 172).`
 
Using the pylon viewer or the C++ code I am able to take images without any issues.
 
For another project I am using a Nvidia Xavier AGX without the OS flashed on an SSD and here the pypylon SDK is working.

`from pypylon import pylon

for i in pylon.TlFactory.GetInstance().EnumerateDevices():
    print("connected camera sid", i.GetSerialNumber())
        
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()
print(cam.GetDeviceInfo)
cam.Close()`

I also tried installing different pylon, pypylon and python versions but it is always giving the same NULL pointer dereferenced error.

I am wondering if booting from the SSD is causing this issue and if there is a possible solution for this?

Thanks

thiesmoeller, 2022-10-26:
can you please start your script in gdb and show the backtrace:

Simonm952, 2022-10-27:
Thanks @thiesmoeller for the reply.

This is the backtrace:

#0  0x0000007fb5e8b648 in std::istream::read(char*, long) () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#1  0x0000007fb5bd7848 in GenApi_3_1_Basler_pylon::CProperty::FromFile(std::istream&) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libNodeMapData_gcc_v3_1_Basler_pylon.so
#2  0x0000007fb61606d8 in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libGenApi_gcc_v3_1_Basler_pylon.so
#3  0x0000007fb6160da8 in GenApi_3_1_Basler_pylon::CNodeMapFactory::CNodeMapFactoryImpl::CacheRead(unsigned int, GenICam_3_1_Basler_pylon::gcstring, GenApi_3_1_Basler_pylon::INodeMapPrivate**, GenApi_3_1_Basler_pylon::CLock*) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libGenApi_gcc_v3_1_Basler_pylon.so
#4  0x0000007fb6161a48 in GenApi_3_1_Basler_pylon::CNodeMapFactory::CNodeMapFactoryImpl::Preprocess(GenICam_3_1_Basler_pylon::gcstring, GenApi_3_1_Basler_pylon::INodeMapPrivate**, GenApi_3_1_Basler_pylon::CLock*) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libGenApi_gcc_v3_1_Basler_pylon.so
#5  0x0000007fb6163048 in GenApi_3_1_Basler_pylon::CNodeMapFactory::CNodeMapFactoryImpl::CreateNodeMap(GenICam_3_1_Basler_pylon::gcstring const&, GenApi_3_1_Basler_pylon::CLock*, bool) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libGenApi_gcc_v3_1_Basler_pylon.so
#6  0x0000007faac836c0 in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylon_TL_usb-6.1.0.so
#7  0x0000007faac8386c in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylon_TL_usb-6.1.0.so
#8  0x0000007faac7648c in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylon_TL_usb-6.1.0.so
#9  0x0000007faac93448 in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylon_TL_usb-6.1.0.so
#10 0x0000007faac947d8 in ?? () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylon_TL_usb-6.1.0.so
#11 0x0000007fb6a4df18 in Pylon::CTlFactory::InternalCreateDevice(Pylon::CDeviceInfo const&, GenICam_3_1_Basler_pylon::gcstring_vector const&, bool) () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylonbase-6.1.0.so
#12 0x0000007fb6a4dd6c in Pylon::CTlFactory::CreateDevice(Pylon::CDeviceInfo const&) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylonbase-6.1.0.so
#13 0x0000007fb6a4e0d8 in Pylon::CTlFactory::InternalCreateDevice(Pylon::CDeviceInfo const&, GenICam_3_1_Basler_pylon::gcstring_vector const&, bool) () from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylonbase-6.1.0.so
#14 0x0000007fb6a4de14 in Pylon::CTlFactory::CreateFirstDevice(Pylon::CDeviceInfo const&) ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/libpylonbase-6.1.0.so
#15 0x0000007fb7684d84 in _wrap_TlFactory_CreateFirstDevice ()
   from /home/simon/.local/lib/python3.7/site-packages/pypylon/_pylon.cpython-37m-aarch64-linux-gnu.so
#16 0x00000000005a7204 in cfunction_call_varargs (kwargs=<optimized out>, args=<optimized out>,
    func=<built-in method TlFactory_CreateFirstDevice of module object at remote 0x7fb7706d70>) at ../Objects/call.c:772
#17 PyCFunction_Call (func=<built-in method TlFactory_CreateFirstDevice of module object at remote 0x7fb7706d70>,
    args=<optimized out>, kwargs=<optimized out>) at ../Objects/call.c:788
#18 0x0000000000513300 in do_call_core (kwdict=0x0,
    callargs=(<TlFactory(this=<SwigPyObject at remote 0x7fb791d2d0>) at remote 0x7fb782ef90>,),
    func=<built-in method TlFactory_CreateFirstDevice of module object at remote 0x7fb7706d70>) at ../Python/ceval.c:4641
#19 _PyEval_EvalFrameDefault (
    f=f@entry=Frame 0x7fb7817c90, for file /home/simon/.local/lib/python3.7/site-packages/pypylon/pylon.py, line 1582, in CreateFirstDevice (self=<TlFactory(this=<SwigPyObject at remote 0x7fb791d2d0>) at remote 0x7fb782ef90>, args=()), throwflag=throwflag@entry=0)
    at ../Python/ceval.c:3191
#20 0x000000000050b47c in PyEval_EvalFrameEx (throwflag=0,
    f=Frame 0x7fb7817c90, for file /home/simon/.local/lib/python3.7/site-packages/pypylon/pylon.py, line 1582, in CreateFirstDevice (self=<TlFactory(this=<SwigPyObject at remote 0x7fb791d2d0>) at remote 0x7fb782ef90>, args=())) at ../Python/ceval.c:547
#21 _PyEval_EvalCodeWithName (_co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>,
    argcount=<optimized out>, kwnames=<optimized out>, kwargs=<optimized out>, kwcount=<optimized out>, kwstep=1, defs=0x0,
    defcount=0, kwdefs=0x0, closure=0x0, name='CreateFirstDevice', qualname='TlFactory.CreateFirstDevice') at ../Python/ceval.c:3930
#22 0x00000000005a77c8 in _PyFunction_FastCallKeywords (func=<optimized out>, stack=0x7fb78bd5d0, nargs=1, kwnames=<optimized out>)
    at ../Objects/call.c:425
#23 0x000000000050e69c in call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=0x7fffffee40) at ../Python/ceval.c:4616
#24 _PyEval_EvalFrameDefault (
    f=f@entry=Frame 0x7fb78bd450, for file /home/simon/Documents/Jetson-OPCUA-Communication-master/Client_application/camera.py, line 8, in <module> (), throwflag=throwflag@entry=0) at ../Python/ceval.c:3110
#25 0x000000000050b47c in PyEval_EvalFrameEx (throwflag=0,
    f=Frame 0x7fb78bd450, for file /home/simon/Documents/Jetson-OPCUA-Communication-master/Client_application/camera.py, line 8, in <module> ()) at ../Python/ceval.c:547
[... cut ...]
#35 pymain_main (pymain=<optimized out>, pymain=<optimized out>) at ../Modules/main.c:3028
#36 0x0000000000649b00 in _Py_UnixMain (argc=<optimized out>, argv=<optimized out>) at ../Modules/main.c:3063
#37 0x0000007fb7e787a0 in __libc_start_main (main=0x0, argc=0, argv=0x0, init=<optimized out>, fini=<optimized out>,
    rtld_fini=<optimized out>, stack_end=<optimized out>) at ../csu/libc-start.c:310
#38 0x00000000005aef1c in _start () at ../Python/pystrtod.c:351
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

thiesmoeller, 2022-10-29:
One reason could be a somewhat corrupt binary cache file in the folder `$HOME/genicam_xml_cache`

can you please try a 
```bash
rm -r $HOME/genicam_xml_cache
```

Simonm952, 2022-11-02:
This solved the issue, thanks!

closed 2022-11-02 by thiesmoeller