basler/pypylon #546: Cameras defaulting to usb 2.0

matkir, 2022-11-07:
Hi,

When working with some daA3840-45uc usb cameras i see that the cameras sometimes defaults to the usb2.0 protocol.

1. When the code is initialized the resulting fps for the cameras is ~10 fps
2. After some time, or when the code crashes unexpectedly, the cameras initializes to ~1 fps
3. restarting the program does not fix the framerate/bandwidth, but physically removing the camera does.  

We have found out that the easiest way to fix this problem is to open pylonviewer and software reset the camera, but this is a manual process we want to automate.
Is there a way to programmatically reset the cameras with pypylon, or is there anywhere we can find documentation on the problem.

thiesmoeller, 2022-11-07:
If Software reset is pushing DeviceReset button in Pylon Viewer...
It is as simple as calling

```python
...
cam.DeviceReset.Execute()
...
```

But this is more a sign of a really instable electrical connection. 
The decision to operate in SuperSpeed is done during link negotiation in the plug and play process of the operating system drivers. Outside the control of Pylon.
Please check your cables and or  host controller connection ( on a PC e.g. use backside ports )

SMA2016a, 2022-11-07:
Try this : 
camera.DeviceReset.Execute();