basler/pypylon #547: Stop received immediately after StartGrabbing

jsiebert, 2022-11-08:
Hey there,

I'm running a system with eight GigE Cameras which grabs images using a hardware trigger (only one frame per trigger). The system is required to run 24/7 but there might be some phases where it is idle, for example during the night. The system runs on Ubuntu 20.04 LTS with Python 3.8.10 and PyPylon 1.8.0.

When I run my code and leave the system without triggering the hardware trigger for some time, it receives ```OnCameraDeviceRemoved``` events for single cameras. Because I couldn't get rid of this, I implemented a functionality to restart all cameras by calling ```camera_array.DestroyDevice()``` (also tried ```Close()``` and ```DetachDevice()```). If all cameras from the ```camera_array```have been reported as 'Destroyed' (or 'Closed' or 'Detached') the software reconnects them and calls ```camera_array.StartGrabbing()``` again.

Now, when calling ```camera_array.StartGrabbing()```, in some cases it could happen that, right after the grabbing has started, it is stopped again immediately (see end of log below):

```
2022-11-08 08:57:11,832 OnCameraDeviceRemoved event for device c1
2022-11-08 08:57:11,836 Camera c1 reported offline. Restarting cameras...

2022-11-08 08:57:11,837 OnGrabStop event for device: c7
2022-11-08 08:57:11,861 OnGrabStopped event for device: c7
2022-11-08 08:57:11,862 OnGrabStop event for device: c8
2022-11-08 08:57:11,901 OnGrabStopped event for device: c8
2022-11-08 08:57:11,902 OnGrabStop event for device: c3
2022-11-08 08:57:11,941 OnGrabStopped event for device: c3
2022-11-08 08:57:11,942 OnGrabStop event for device: c5
2022-11-08 08:57:11,981 OnGrabStopped event for device: c5
2022-11-08 08:57:11,982 OnGrabStop event for device: c2
2022-11-08 08:57:12,022 OnGrabStopped event for device: c2
2022-11-08 08:57:12,022 OnGrabStop event for device: c1
2022-11-08 08:57:12,061 OnGrabStopped event for device: c1
2022-11-08 08:57:12,062 OnGrabStop event for device: c4
2022-11-08 08:57:12,102 OnGrabStopped event for device: c4
2022-11-08 08:57:12,103 OnGrabStop event for device: c6
2022-11-08 08:57:12,142 OnGrabStopped event for device: c6

2022-11-08 08:57:12,143 OnClose event for device: c7
2022-11-08 08:57:12,152 OnClosed event for device: c7
2022-11-08 08:57:12,152 OnDestroy event for device: c7
2022-11-08 08:57:12,180 OnDestroyed event for device: c7

2022-11-08 08:57:12,180 OnClose event for device: c8
2022-11-08 08:57:12,183 OnClosed event for device: c8
2022-11-08 08:57:12,183 OnDestroy event for device: c8
2022-11-08 08:57:12,192 OnDestroyed event for device: c8

2022-11-08 08:57:12,192 OnClose event for device: c3
2022-11-08 08:57:12,194 OnClosed event for device: c3
2022-11-08 08:57:12,194 OnDestroy event for device: c3
2022-11-08 08:57:12,201 OnDestroyed event for device: c3

2022-11-08 08:57:12,201 OnClose event for device: c5
2022-11-08 08:57:12,203 OnClosed event for device: c5
2022-11-08 08:57:12,203 OnDestroy event for device: c5
2022-11-08 08:57:12,210 OnDestroyed event for device: c5

2022-11-08 08:57:12,210 OnClose event for device: c2
2022-11-08 08:57:12,212 OnClosed event for device: c2
2022-11-08 08:57:12,212 OnDestroy event for device: c2
2022-11-08 08:57:12,218 OnDestroyed event for device: c2

2022-11-08 08:57:12,218 OnClose event for device: c1
2022-11-08 08:57:12,220 OnClosed event for device: c1
2022-11-08 08:57:12,220 OnDestroy event for device: c1
2022-11-08 08:57:12,226 OnDestroyed event for device: c1

2022-11-08 08:57:12,226 OnClose event for device: c4
2022-11-08 08:57:12,228 OnClosed event for device: c4
2022-11-08 08:57:12,228 OnDestroy event for device: c4
2022-11-08 08:57:12,235 OnDestroyed event for device: c4

2022-11-08 08:57:12,235 OnClose event for device: c6
2022-11-08 08:57:12,237 OnClosed event for device: c6
2022-11-08 08:57:12,237 OnDestroy event for device: c6
2022-11-08 08:57:12,244 OnDestroyed event for device: c6

2022-11-08 08:57:12,244 Waiting for cameras to come online again...
[... 92 lines cut ...]

   if len(devices) == 0:
      logging.error("No cameras found! Check hardware installation!")
   else:
      self.cameras = pylon.InstantCameraArray(min(len(devices), len(self.config["cameras"])))
      for i, cam in enumerate(self.cameras):
         cam.RegisterConfiguration(CameraEventHandler(self.__device_offline, self.__device_destroyed), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
         cam.RegisterImageEventHandler(ImageEventHandler(config, self.image_data_collector), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
         cam.Attach(tl_factory.CreateDevice(devices[i]))
         cam.Open()

except genicam.GenericException as e:
   logging.error(f"An error occurred during camera initialization: {e}!")
```

```StartGrabbing``` is called as follows:
```python
self.cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
```

Does anybody have an idea on this or maybe seen the same behaviour and probably solved this?

Thanks in advance,

Jens

thiesmoeller, 2022-11-08:
Looking a reasons why your cameras disappear.

If there is no active communication with the cameras, a background heartbeat protocol is run between your PC and the cameras.

If the PC is not receiving an answer ( after retries ) the camera is declared lost and you get the result described.

In your night time idle situation the CPU is definitely in higher power saving c states on most cores and will need a significant time to properly wake up.

Have your already applied the network optimization settings as described in the networking sections on docs.baslerweb.com ?
This will increase the number of receive descriptors on the network interface.

One additional check could also be to disable higher c states for the whole system ( aggressive but a good check )
This can be done with kernel command line parameters

jsiebert, 2022-11-09:
Hey @thiesmoeller,

thanks for the reply! I'm pretty sure we've done everything mentioned in the Basler documentation regarding network configuration, with the exception of disabling c-states. But I'll check with the hardware folks....

However, this doesn't solve the general problem that restarting the camera/the grabbing sometimes doesn't work properly. We use this feature for other use cases as well, not just the case mentioned above.

Do you have any idea for this problem? If not a solution, any advice on how to debug the problem would be greatly appreciated!

thiesmoeller, 2022-11-11:
Would recommend to approach issue from two sides.

A) understand what happens in your system when the cameras disappear. 
 First tool could be record GVCP traffic from all cameras with Wireshark 24/7 in loop mode and stop the Wireshark process in case of your error situation.

B) Run pypylon with enabled logging. If you install full pylon SDK there is a start-with-logging script. Has to be adapted slightly to use it with python as executable.
This will give you more insight in what is happening in case of error

jsiebert, 2022-11-14:
Running pypylon with logging enabled did the trick! The following error message occurred in the log right before the grabbing was stopped:

```shell
[2022-11-14 09:40:33,013:pylon.base.waitobject] ERROR: . Reached open files limit: To change the limit permanently on a ubuntu system, edit /etc/security/limits.conf
```

After setting the open file limit to "unlimited" the problem was solved!

```shell
ulimit -Hn
1048576

ulimit -Sn
1048576
```

Full log attached for reference, closing this issue... Thanks @thiesmoeller for your help! 👍 

[pylonLog.txt](https://github.com/basler/pypylon/files/10001553/pylonLog.txt)

thiesmoeller, 2022-11-14:
Did this also fix loosing the camera in the first place.

-> then I would really recommend to check which process is using so many file handles ( or even leaking them)

jsiebert, 2022-11-14:
> Did this also fix loosing the camera in the first place.

Not sure about this yet... Will observe this over the next few days...

> 
> -> then I would really recommend to check which process is using so many file handles ( or even leaking them)

I did an ```strace``` on my application and the only thing that I can see is that there are many file opened from the ```genicam_xml_cache```folder on the home directory. But they never seem to be closed. 

```shell
[...]
[pid 31908] stat("config/seifert/acA3088-16gm.pfs", {st_mode=S_IFREG|0664, st_size=37779, ...}) = 0
[pid 31908] openat(AT_FDCWD, "config/seifert/acA3088-16gm.pfs", O_RDONLY) = 829
[pid 31908] openat(AT_FDCWD, "/dev/shm/sem.168b266c917d762355GenICam_XML", O_RDWR|O_NOFOLLOW) = 829
[pid 31908] access("/home/doks/genicam_xml_cache/0x1936748300000000.bin", F_OK) = 0
[pid 31908] openat(AT_FDCWD, "/home/doks/genicam_xml_cache/0x1936748300000000.bin", O_RDONLY) = 829
[pid 31908] openat(AT_FDCWD, "/dev/shm/sem.168b266a100beef973GenICam_XML", O_RDWR|O_NOFOLLOW) = 829
[pid 31908] access("/home/doks/genicam_xml_cache/0xd087917900000000.bin", F_OK) = 0
[pid 31908] openat(AT_FDCWD, "/home/doks/genicam_xml_cache/0xd087917900000000.bin", O_RDONLY) = 829
[pid 31908] openat(AT_FDCWD, "/proc/sys/net/core/rmem_max", O_RDONLY) = 835
[pid 31908] openat(AT_FDCWD, "/dev/shm/sem.1570d298148e8c7b44GenICam_XML", O_RDWR|O_NOFOLLOW) = 835
[pid 31908] access("/home/doks/genicam_xml_cache/0xe4d75b2a00000000.bin", F_OK) = 0
[pid 31908] openat(AT_FDCWD, "/home/doks/genicam_xml_cache/0xe4d75b2a00000000.bin", O_RDONLY) = 835
[pid 31908] openat(AT_FDCWD, "/dev/shm/sem.168b2644c339964413GenICam_XML", O_RDWR|O_NOFOLLOW) = 837
[pid 31908] access("/home/doks/genicam_xml_cache/0xbf345cc000000000.bin", F_OK) = 0
[pid 31908] openat(AT_FDCWD, "/home/doks/genicam_xml_cache/0xbf345cc000000000.bin", O_RDONLY) = 837
[...]
```

I guess these calls are the result from calling 

```python
pylon.FeaturePersistence.Load(config_file, cam.GetNodeMap())
```

for each configured camera in the system. Full log attached for reference...
[strace.log](https://github.com/basler/pypylon/files/10002344/strace.log)

closed 2022-12-16 by jsiebert