basler/pypylon #548: Hardware triggering

sajidaarif, 2022-11-09:
Hi All

I am trying to hardware trigger a Basler camera. I have located the GIPO lines and the ground on the camera . I seem to have setup everything correctly but still can't get to start image acquisition when I send the signal from an Arduino.
I have attached screen shots of Pylon set up. I have been working on it for almost two months now. 
[image]
[image]
[image]
[image]

thiesmoeller, 2022-11-09:
Oh.. two months of trying.

Hope this comment in the issues helps ( also check discussion )
https://github.com/basler/pypylon/issues/470#issuecomment-1178811364

And also my other sample code here:
https://github.com/basler/pypylon-samples/blob/main/notebooks/USB_hardware_trigger_and_chunks.ipynb

And the hw trigger will always only trigger a frame start. You have to start the acquisition before ( StartGrabbing )

thiesmoeller, 2022-11-11:
What to do on Arduino is not our domain... But there are plenty of docs at the math works : https://www.mathworks.com/help/supportpkg/arduinoio/ref/arduino.html

All the code I mentioned is to configure a Basler camera to be triggered by hardware.

You have to generate digital pulses on your Arduino and configure the camera to properly react on signal changes detected on one of its GPIO lines

sajidaarif, 2022-11-14:
Hi

I have already generated the digital signals and confirmed with an oscilloscope, in addition to a two-photon microscope. The only problem is the camera, I just can't trigger it.
Best,
Sajida

thiesmoeller, 2022-12-14:
Have you applied https://github.com/basler/pypylon-samples/blob/main/notebooks/USB_hardware_trigger_and_chunks.ipynb to your setup.

This is like a Logic-Analyzer to show signals on the input of the camera and "debug" the trigger system

chiyu1203, 2023-05-20:
> #470 (comment)

Hi Sajida, Have you managed to trigger the camera with Arduino?
I also want to trigger a basler camera with Arduino UNO. Therefore I would appreciate if you can share with me your experience.
Best 
Chi-Yu

rambo, 2023-06-26:
The opto-isolated trigger input needs about 20mA of current so if triggering multiple cameras from Arduino (or similar) you should have a buffer (I usually use 74hc04 with inputs and outputs connected in parallel). MCUs with 3.3V IO can fail to trigger even one camera.

Or alternative (especially if using something else than arduino with "open collector" style opto-isolated outputs) like so https://www.digikey.fi/schemeit/project/arduino-open-collector-style-drive-540e839d77044ba6b9c2c58871b6352c

In this case you will want to invert the line in camera config (or drive the arduino pin LOW for trigger and HIGH when idle).

thomaswelter, 2024-12-05:
I've also had problems with getting the GPIO input trigger to work. Maybe I'm doing something wrong but I've had this issue with multiple camera's. If your camera has an opto coupled input that usually works. Don't forget to connect the opto couple ground!

HighImp, 2024-12-05:
Hi @thomaswelter, can you please provide your Shematic and your code? For the GPIOs, the Arduino should be able to supply the desired current. To be sure that the voltage level is being handled in the camera, you can check the line status: https://docs.baslerweb.com/line-status. If this behaves as desired, this could be an indication of the cause, may its the trigger configuration afterwards.