basler/pypylon #549: Hardware Trigger

albertovitto, 2022-11-12:
Hi,
I'm trying to convert the following code using a background loop. The camera stays on 24/7 and has to wait the signal of an hardware trigger before grabbing a frame; an external function will process each of them. The number of frames to grab is undefined.
I've tried different solutions proposed in
[https://github.com/basler/pypylon/blob/master/samples/grabstrategies.py](url), 
[https://github.com/basler/pypylon/blob/master/samples/grabcameraevents.py](url), 
[https://github.com/basler/pypylon-samples/blob/main/notebooks/USB_hardware_trigger_and_chunks.ipynb](url), and 
[https://github.com/ridvanozdemir/hardware_trigger_basler_camera/blob/main/basler_hardware_trigger.py](url)
but none of them seems to work and I don't understand why.
Is there a better way to do that or my code just looks fine?

Thanks, any help would be appreciated!

```
import pypylon.pylon as py
import time

    tlf = py.TlFactory.GetInstance()

    camera = py.InstantCamera()

    camera.Attach(tlf.CreateFirstDevice())

    camera.Open()

    # set hardware trigger
    camera.ChunkModeActive = True
    camera.ChunkSelector = "LineStatusAll"
    camera.ChunkEnable = True
    camera.MaxNumBuffer = 10

    camera.UserSetSelector = "Default"
    camera.UserSetLoad.Execute()

    camera.TriggerSelector = "FrameStart"
    camera.TriggerSource = "Line1"
    camera.TriggerActivation = 'RisingEdge'
    camera.TriggerMode = "On"

    camera.StartGrabbing()

    count = 0

    while True:
        with camera.RetrieveResult(py.waitForever, py.TimeoutHandling_ThrowException) as res:

            if res.GrabSucceeded():
                img = res.GetArray()
                print(f"processing image #{count}..")
                # process_frame(img)
                time.sleep(3)
                count += 1
            else:
                print("error while grabbing..")
                break

    camera.StopGrabbing() 
    camera.Close()
```

SMA2016a, 2022-11-17:
refer the following sample code 
https://github.com/basler/pypylon/blob/master/samples/grabstrategies.py
you can use the callback onimage() to get informed when an image is ready.

albertovitto, 2022-11-26:
> 

Yes I've tried following that sample but I'm not able to refactor the code according to my needs.
Could you please help me?

niallmulkerns, 2023-04-14:
Replying quite late, so no idea if you're still in need of the code, but I managed to get the following code taken from a sample above to work on my system using a 5V p-p signal on pins 6 and 3 (ground + GPIO, [https://docs.baslerweb.com/aca1920-40um](https://docs.baslerweb.com/aca1920-40um)). No doubt much of it is redundant, but it works! I find that on my system, the cv window won't trigger unless I make the wait time much longer than the pulse period for some reason - I guess there are overheads? 

    import time
    import cv2
    from pypylon import pylon
    
    total_time = time.time()
    
    tl_factory = pylon.TlFactory.GetInstance()
    camera = pylon.InstantCamera()
    img = pylon.PylonImage()
    camera = pylon.InstantCamera(tl_factory.CreateFirstDevice())
    
    
    print('Start acquiring')
    camera.Attach(tl_factory.CreateFirstDevice())
    camera.Open()
    
    #camera settings for triggering an input
    camera.MaxNumBuffer = 1
    camera.LineSelector = "Line4"
    camera.LineMode = "Input"
    camera.TriggerSource ='Line4'
    camera.TriggerSelector ='FrameStart'
    camera.TriggerMode ='On'
    camera.TriggerActivation='RisingEdge'
    camera.AcquisitionMode='Continuous'
    
    print("Trigger mode: ", camera.TriggerMode.Value)
    
    converter = pylon.ImageFormatConverter()
    
    # converting to opencv bgr format
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
    
    class ImageEventPrinter(pylon.ImageEventHandler):
        def OnImagesSkipped(self, camera, countOfSkippedImages):
            print("OnImagesSkipped event for device ", camera.GetDeviceInfo().GetModelName())
            print(countOfSkippedImages, " images have been skipped.")
    
        def OnImageGrabbed(self, camera, grabResult):
            print("OnImageGrabbed event for device ", camera.GetDeviceInfo().GetModelName())
            print('Waiting for trigger')
            if grabResult.GrabSucceeded():
                print("SizeX: ", grabResult.GetWidth())
                print("SizeY: ", grabResult.GetHeight())
                image = converter.Convert(grabResult)
                image = image.GetArray()
                print("image taken")
                cv2.namedWindow("Detection Window", cv2.WINDOW_NORMAL)
                cv2.resizeWindow("Detection Window", 1024,1024)
                cv2.imshow('Detection Window', image)
                print(camera.TriggerMode.Value)
                k= cv2.waitKey(1)
                if k == 27:
                    print("program terminated")
                    grabResult.Release()
                    camera.StopGrabbing()
                    cv2.destroyAllWindows()
                    camera.Close()
    
    camera.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)
    
    while time.time() - total_time < 2:
        # camera.LineSelector = "Line4"
        print(camera.LineStatus.Value)
    
    
    print("total time: ", (time.time() - total_time))
    camera.StopGrabbing()