basler/pypylon #550: pylon.TlFactory.GetInstance() Error

naubf, 2022-11-14:
Hello everyone,
I am trying to create a program by using PyPylon and Basler acA1920-40gc.
At the first attempt the camera could be connected only to ended up with blue screen error several seconds after that.
And after rebooting the PC, Pypylon wouldn't let me connect to the camera at all.

The error happened at this line:
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
or
devices = pylon.TlFactory.GetInstance().EnumerateDevices()

When the program done on .Py it shows this result:
Assertion failed: (false) && "Failed to load embedded XML", file d:\jenkinscore\workspace\pylon-release_release_6.3@2\pylon\resourcehelper\embeddednodemapfactory.cpp, line 44

While done on .ipynb:
Canceled future for execute_request message before replies were done
The Kernel crashed while executing code in the the current cell or a previous cell. Please review the code in the cell(s) to identify a possible cause of the failure. Click here for more info. View Jupyter log for further details.

Windows 10 64bit
Python 3.7, 3.8, 3.9 and 3.10 show the same result.
PyPylon installed from pip and also github, both show same result.

Has anybody else encountered similar issues or know how to solve it, can you help me to solve this?

Thank you.

thiesmoeller, 2022-11-14:
Is the pylonviewer running correctly on your system in this configuration ?

naubf, 2022-11-14:
@thiesmoeller Yes, I can use the camera with no problem with pylonviewer.

naubf, 2022-11-15:
Already tried another attempt through Pylonviewer. Single image capture can be done without an error while capturing video ended up with bluescreen error again. (Error Code: DPC_WATCHDOG_VIOLATION)