basler/pypylon #553: pylon image reset.

sulaimansiddiqi233, 2022-11-15:
hello.

we just want to reset the image by using result.image.reset function but it not works currently . 
could check and help us. thank you.

[image]
[image]

sulaimansiddiqi233, 2022-11-19:
@thiesmoeller @SMA2016a

SMA2016a, 2022-11-20:
is the result variable member of vTools recipe?  if so, did you connect output pin?  why do you call reset before ? 

do you want to resize the image?

Would you send minimum code  to reproduce the effect?

sulaimansiddiqi233, 2022-11-20:
yes we use recipe and want to resize image after we get the barcode result.

/*
    Sample using the Barcode vTool (license required)
*/
// Include files to use the pylon API.
#define BUILD_DLL
#include <pylon/PylonIncludes.h>
#ifdef PYLON_WIN_BUILD
#  include <pylon/PylonGUI.h>
#endif
// Extend the pylon API for using pylon data processing.
#include <pylondataprocessing/PylonDataProcessingIncludes.h>
// The sample uses the std::list.
#include <list>
#include <iostream>
#include <fstream>
#include <stdlib.h>
#include <stdio.h>
#include <mysql.h>
#include <string> 
#include <windows.h>
#include <pylon/ImagePersistence.h>

// Namespaces for using pylon objects
using namespace Pylon;
using namespace Pylon::DataProcessing;

// Namespace for using cout
using namespace std;

// Number of images to be grabbed
static const uint32_t c_countOfImagesToGrab = 100;

// Declare a data class for one set of output data values.
class ResultData
{
public:
    ResultData()
        : hasError(false)
    {
    }

    CPylonImage image; // The image from the camera

    StringList_t decodedBarcodes; // The decoded barcodes

    bool hasError;     // If something doesn't work as expected
    // while processing data, this is set to true.

    String_t errorMessage; // Contains an error message if
    // hasError has been set to true.
};

// MyOutputObserver is a helper object that shows how to handle output data
// provided via the IOutputObserver::OutputDataPush interface method.
// Also, MyOutputObserver shows how a thread-safe queue can be implemented
// for later processing while pulling the output data.

class MyOutputObserver : public IOutputObserver
{
public:
    MyOutputObserver()
        : m_waitObject(WaitObjectEx::Create())
    {
    }

    // Implements IOutputObserver::OutputDataPush.
    // This method is called when an output of the CRecipe pushes data out.
    // The call of the method can be performed by any thread of the thread pool of the recipe.
[... 233 lines cut ...]
            }
            else
            {
                throw RUNTIME_EXCEPTION("Result timeout");
            }
        }

        // Stop the processing.
        recipe.Stop();
    }
    catch (const GenericException& e)
    {
        // Error handling
        cerr << "An exception occurred." << endl << e.GetDescription() << endl;
        exitCode = 1;
    }

    // Comment the following two lines to disable waiting on exit.
    cerr << endl << "Press enter to exit." << endl;
    while (cin.get() != '\n');

    // Releases all pylon resources.
    PylonTerminate(true);
    return exitCode;
`}`

SMA2016a, 2022-11-21:
The purpose of the function is not for  resizing image, instead of resizing allocated buffer of the image.

closed 2022-11-21 by sulaimansiddiqi233