basler/pypylon #554: GigE Decompression memory leak

lasjoe, 2022-11-17:
Hello,

I'm having issues setting up GigE camera with decompression, it does work but results in significant memory leak.

I'm probably missing something obvious here, any help would be appreciated.

Thanks in advance

Code:

```
from pypylon import pylon
import time
import cv2

ip_address = 'iphere'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

camera.Open()

camera.ImageCompressionMode = "BaslerCompressionBeyond"
camera.ImageCompressionRateOption = "Lossless"
camera.BslImageCompressionRatio = 100.0

decompressor = pylon.ImageDecompressor()
descriptor = camera.BslImageCompressionBCBDescriptor.GetAll()
decompressor.SetCompressionDescriptor(descriptor)

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
framecounter = 0
while(camera.IsGrabbing):
    with camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as payload:
        try:
            decompressed_image = decompressor.DecompressImage(payload)
            decompressed_array = decompressed_image.Array
            print(framecounter)
            framecounter += 1
        except:
            print("Failed to decompress payload")
```

[image]

SMA2016a, 2022-11-17:
you may try to release the grab result, e.g payload

lasjoe, 2022-11-17:
Sadly same issues with

```
while(camera.IsGrabbing):
    with camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as payload:
        try:
            decompressed_image = decompressor.DecompressImage(payload)
            decompressed_array = decompressed_image.Array
            print(framecounter)
            framecounter += 1
        except:
            print("Failed to decompress payload")
        payload.Release()
```

SMA2016a, 2022-11-18:
what about decompressed_image? could it also be released?

lasjoe, 2022-11-18:
I tried that too, no luck, tested on multiple machines, Windows 10 and Server 2019
```
while(camera.IsGrabbing):
    with camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as payload:
        try:
            decompressed_image = decompressor.DecompressImage(payload)
            decompressed_array = decompressed_image.Array
            print(framecounter)
            framecounter += 1
            decompressed_image.Release()
        except:
            print("Failed to decompress payload")
        payload.Release()
```

SMA2016a, 2022-11-18:
yes, I am also seeing this effect. If I find anything more come back to with the result

thiesmoeller, 2023-02-03:
Hi @lasjoe 

we've create a branch that fixes the memory leak.

See https://github.com/basler/pypylon/tree/feature/fix_image_decompress_leak

Full mainline release via with pip packages via pypi will follow.

lasjoe, 2023-05-25:
Thanks for fixing it so quickly @thiesmoeller

I just did a quick test and it seems memory usage doesn't increase when decompression errors occur.
I'll do more thorought tests with one more camera next week.

closed 2023-05-31 by stefanklug