basler/pypylon #555: Error when installing  Camera Enablement Package for NVIDIA®’s Jetson boards

visionad, 2022-11-17:
When i try to install this package on my jetson nano i have the following error:
[image]

On that list of hardware modules it should appear: 
 1. Adafruit SPH0645LM4H
  2. Basler camera overlay
  3. FE-PI Audio Z V2
But as you can see, the basler camera overley is missing. 

I have tried both the installation both with the setup.py automatic installation and the manual installation. The error appears in both, when it checks that the overly is available. 

The manual installtion guide says: 
_2. Ensure that the overlay is available by running:

    ```sh
    $ sudo /opt/nvidia/jetson-io/config-by-hardware.py -l
    Configurations for the following hardware modules are available:
    
    1. Adafruit SPH0645LM4H
    2. Basler camera overlay
    3. FE-PI Audio Z V2
    ```_

Im using a Jetson Nano with Jetpack 4.6.1-b110

I cant find any similar error on the internet, 

Thank you in advanced

SMA2016a, 2022-11-17:
do you need to have the latest CEP Version.   Contact your local distributor

visionad, 2022-11-17:
Thank you for responding,

As far as i know, there is only one CEP Version for Jetson, so i dont have the possibility to upgrade the CEP,

Is there anything else i could do?

visionad, 2022-11-18:
Solved,

In the tegra... .dts file, between the overley name and compatible lines it needs to be this line:
jetson-header-name = "Jetson 40pin Header"

Like this:

[image]

closed 2022-11-23 by visionad