basler/pypylon #557: use pointer to buffer not cam.RetrieveResult

SpangeJ, 2022-11-23:
Hello,
My SW is multithreaded where:
One thread does image acquisition and appends metadata from other sensors (accelerometer ++)  and pushes it all to a queue
Another thread pops from the queue and does a computation on the images + metadata

My issue is that the command
`cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)` takes ~100ms pr. camera and that is when images are already in the buffer `(cam.NumReadyBuffers() > 0)`.

My Idea of speeding up my code is to move the `RetriveResult` from my acquisition thread to my processing thread.
I am thinking that my Acquisition queue should contain pointers to the memory address and not the retrieved result itself
By doing something like:

> Acquiction thread: 
&nbsp; queu_element = type('queue_element', (object,), {})()
&nbsp; pointers = [None]*len(cameras)
&nbsp; for i, cam in enumerate(cameras):
&nbsp; &nbsp; while cam.NumReadyBuffers() == 0:
&nbsp; &nbsp; &nbsp; sleep(0.01)
&nbsp; &nbsp; pointers[i] = cam.PointerToBufferedDataNr(0)
&nbsp; queue_element.pointers = pointers
&nbsp; queue_element.metadata = get_metadata()
&nbsp; queue.push(queue_element)

> Processing thread:
&nbsp; queue_element = queue.pop()
&nbsp; Images = [None] * len(queue_element.pointers)
&nbsp; for i, pointer in enumerate(queue_element.pointers)
&nbsp; &nbsp; Images[i] = RetrieveResultFromMemoryAddess(5000, pylon.TimeoutHandling_ThrowException)
&nbsp; do_awsome_processing(Images, queue_element.metadata)

I hope someone knows a way to help me improve my SW

SMA2016a, 2022-11-23:
retrieved result is alraey a point to a data package that contains an image buffer and its property

SpangeJ, 2022-11-24:
If it is a pointer, then why does the command
`grab_result = cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)`
take 100ms to complete?

thiesmoeller, 2022-11-24:
Please take a look at the grab strategies sample in the samples folder or in the companion repository github.com/Basler/pypylon-samples

Your use case is best served with retrieving the images in the pypylon background loop. pypylon will directly call your method without blocking your thread

SpangeJ, 2022-11-25:
@thiesmoeller 
> Please take a look at the grab strategies sample in the samples folder or in the companion repository github.com/Basler/pypylon-samples
> 
> Your use case is best served with retrieving the images in the pypylon background loop. pypylon will directly call your method without blocking your thread

Are you referring to to `camera.GetGrabResultWaitObject().Wait(0)`  which is found in [grabstrategies.py](https://github.com/basler/pypylon/blob/master/samples/grabstrategies.py)?

thiesmoeller, 2022-11-25:
No. Please check the background loop support. 
Which allows to handle the image receive in a callback.

One example is in the section 'Background Loop' in https://github.com/basler/pypylon-samples/blob/c3e323c07b0e0efaf59a85685d35ff36056d2ef9/notebooks/grabstrategies.ipynb

SpangeJ, 2022-11-29:
An example scenario of what I try to accomplish would be to send 10x external trigger signals with 100ms gap to three cameras simultaneously. When a new image appears in memory (every 100ms) I would like to read the most recent available values from my external sensors and append them to the image data.
Ideally, I would send the following structure to my queue, to be picked up by the processing thread:
`queue_element.pointer0 = pointer to the most recent image in the buffer from camera 0`
`queue_element.pointer1 = pointer to the most recent image in the buffer from camera 1`
`queue_element.pointer2 = pointer to the most recent image in the buffer from camera 2`
`queue_element.metadata = get_last_reading_from_external_sensors()`

@thiesmoeller, I am trying to understand your `Background Loop` I can run the code locally with one camera in Jupyter.
I'm having a tough time implementing your example into my code.
I have multiple cameras and I use an external triggering signal; do you have any suggestions?
Currently, I'm struck with `cam.NumReadyBuffers.GetValue()` always being zero.

thiesmoeller, 2022-12-02:
One question before providing answer: how do you currently measure the 100ms latency  you detect?

SpangeJ, 2022-12-06:
I measure the 100ms the following way:

> while cam.NumReadyBuffers.GetValue() == 0:
&nbsp; time.sleep(0.01)
Now = time.time()
grab_result = cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
print(f"Grab took: {time.time() - NOW}ms")

The measured time varies a lot, but at its peak maximum, it can be as much as 100ms.
And, since I have three cameras being triggered simultaneously by an external trigger signal, the total acquisition time can be 300ms.
Therefore, any external trigger signal faster than 3.3fps will be faster than python is able to grab the result.
My goal is to achieve 10 fps.

SMA2016a, 2022-12-08:
would you print the camera parameter resultingframerate, please?

SpangeJ, 2022-12-09:
@SMA2016a 

I write:
`print(f"Resulting framerate for cam {i}: {cam.ResultingFrameRate.GetValue()}")`
into my code and get:

[image]

These results surprise me, you see I use an external trigger source (a microcontroller similar to Arduino Uno)
And I've set it to 11 fps.
I would assume all three cameras had the same resulting framerate and something close to 11, but they don't.

NB!
I have set `cam.maxNumBuffer` to something large so images are buffered into the internal queue.
However, this is what I DO NOT WANT because then the sampling of my external sensor data will be lagging behind the camera data.

My core issue is that `grab_result = cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)` takes too long time to run three times (since I'm using three cameras).
That is why I raised the question of not grabbing the image but simply attaching metadata/external sensor data to pointers to the image and then my other queue would pick up the metadata and the three pointers and from those pointers run `RetrieveResult`

SMA2016a, 2022-12-10:
please compare first following parameters of  the cam 2 with camera 1:

Pixelformat
exposure time
if GigE -> Interpacket delay
if USB -> DeviceLinkThroughputLimit  etc.

Easier for you if you open the camera in pylonviewer,  in the left down corner you will find a task selector, select the "controlling the frame rate"

Pylon viewer will highlight all parameters to be compared

Then check with pylon viewer if all cameras are running at 10fps.   if you send your 10Hz signal  without losing any frames

SpangeJ, 2022-12-12:
The pixel format is:
Camera 0 &1: Mono12
Camera 2: Bayer RG12

I did not find the "controlling the frame rate" in the left-down corner (I'm using pylon 6.2.0.21487), so I ran the Bandwith manager:
[image]

Does this answer your question?

SpangeJ, 2022-12-23:
I'll mark this as resolved now.
I'm marking it as resolved because I now know that if I can stream a certain fps in `Bandwidth Manager` in Pylon Viewer then I should be able to receive the same fps in python.

If I for some reason still don't get the expected speed I can:
[Edit USB Steam Maximum Transfer Size](https://github.com/basler/pypylon/issues/253)
Edit [cam.GetStreamGrabberNodeMap().GetNode("MaxTransferSize").Value](https://docs.baslerweb.com/pylonapi/cpp/class_pylon_1_1_c_instant_camera#function-getstreamgrabbernodemap)

closed 2022-12-23 by SpangeJ

dahleri, 2023-04-11:
@SpangeJ I am interested to know how you solved your issue! Did you end up building a queue of pointers with metadata or how do you compare the external sensors to your image?

SpangeJ, 2023-04-13:
Hello @dahleri 
There is a [basler calculator](https://www.baslerweb.com/en/sales-support/tools/frame-rate-calculator/) that can help you know the max fps for a given camera en settings. Typically selecting 12bit resolution results in a slower fps then what's written on the camera. Running the Bandwidth manager I saw that the slowest camera had an fps of 8.63. So I set my external trigger source to ~8.5pfs. Timing my code correctly I verified what @SMA2016a wrote "Retrieved result is already a point to a data package that contains an image buffer and its property". I.E. I was able to run a loop that ran `cam.RetrieveResult` (for all my cameras (I have 3)), collected my metadata (sensor input) and appended to my list/queue and that loop ran at the same fps (8.5), resulting in no delay and no out-of-sync between metadata.
There for I abandoned my initial pointers idea since my code containing `cam.RetrieveResult` was able to run as fast as the images arrived (8.5 fps)