basler/pypylon #558: Set Transmission Via Unicast destination IP and Port

IgnacioEscorz, 2022-11-23:
Hello!

I've been trying to set the transmission type to Unicast, and to set the destination IP and Port, unsuccesfully.
I tried C++ code provided by the Pylon Viewer on WIndows.

camera.GetStreamGrabberParams().TransmissionType.SetValue(TransmissionType_Unicast);
camera.GetStreamGrabberParams().DestinationPort.SetValue(34978);
int64_t i = camera.GetStreamGrabberParams().DestinationPort.GetValue();

The reason for this is that I'm using a Docker that can not be used as host network mode, but I need to have a defined fixed transmission port and use port fordwarding.

Thank you!

IgnacioEscorz, 2022-11-24:
> you need to set TransmissionType_Multicast to be able to set the destination IP.

Thank you @SMA2016a , and what is the code line to change it in Pypylon? I could not find it either.

SMA2016a, 2022-11-25:
try this:

cam.StreamGrabber.TransmissionType ="Multicast"