basler/pypylon #559: Too much buffering (stale frames)

keesjan123, 2022-11-23:
Hi,

I'm having some problems turning off buffering completely. I'm getting stale frames whenever the grab readout loop can't keep up with the incoming frames, which should not be a problem when there is no buffering. In our application (robotics), it does not matter much if frames are dropped, but the frames should be as fresh as possible (so the machine can make a decision based on the video frame data as quickly as possible after e.g. a camera is moved).

I've made a small script (see below) that reproduces the problem.

- Near the top of the script the MaxNumBuffer parameter is set to 1 (in the StreamGrabber interface also).
- The function get_clock_difference() determines the difference between the system clock and the camera clock. We use this to determine the latency of the frames (i.e., the time between first exposure and readout).
- Then a background grab job is started using StartGrabbingMax (I tried this with various parameters; intuitively the grab strategy LatestImageOnly seemed most suited, but it still shows the problem).
- In a grab-loop, we try to remove all pending frames from the buffer (which doesn't work). The last frame is used, and its latency is printed.
- Finally, we wait a bit (100ms) to simulate the case where we can't keep up with the emitted frames; and the frame is released.

The output of the script:

```
Using device  acA1920-40uc
1920x1200 t=0.069676 latency=0.027243 brightness=905
1920x1200 t=0.188586 latency=0.113011 brightness=909
1920x1200 t=0.303065 latency=0.207814 brightness=981
1920x1200 t=0.418386 latency=0.298173 brightness=902
1920x1200 t=0.527313 latency=0.389262 brightness=895
1920x1200 t=0.642215 latency=0.474394 brightness=940
1920x1200 t=0.757873 latency=0.565401 brightness=937
1920x1200 t=0.872842 latency=0.656827 brightness=840
1920x1200 t=0.987171 latency=0.747689 brightness=798
1920x1200 t=1.102810 latency=0.837979 brightness=900
1920x1200 t=1.218132 latency=0.929667 brightness=923
1920x1200 t=1.333814 latency=1.020944 brightness=930
1920x1200 t=1.449307 latency=1.112596 brightness=889
1920x1200 t=1.564973 latency=1.203945 brightness=901
1920x1200 t=1.680538 latency=1.295625 brightness=910
1920x1200 t=1.796282 latency=1.387136 brightness=923
1920x1200 t=1.910552 latency=1.478714 brightness=866
1920x1200 t=2.025930 latency=1.569044 brightness=863
1920x1200 t=2.140132 latency=1.660296 brightness=937
1920x1200 t=2.255595 latency=1.750748 brightness=872
1920x1200 t=2.370761 latency=1.841960 brightness=948
1920x1200 t=2.486417 latency=1.933089 brightness=878
1920x1200 t=2.599303 latency=2.024502 brightness=857
1920x1200 t=2.713834 latency=2.113404 brightness=849
1920x1200 t=2.828306 latency=2.179952 brightness=874
1920x1200 t=2.943103 latency=2.174115 brightness=898
1920x1200 t=3.057371 latency=2.168768 brightness=931
1920x1200 t=3.173227 latency=2.186799 brightness=984
1920x1200 t=3.288033 latency=2.182459 brightness=907
1920x1200 t=3.403662 latency=2.177032 brightness=933
1920x1200 t=3.511864 latency=2.172212 brightness=914
1920x1200 t=3.627487 latency=2.160368 brightness=932
1920x1200 t=3.743147 latency=2.179873 brightness=843
1920x1200 t=3.859803 latency=2.175516 brightness=939
1920x1200 t=3.974153 latency=2.171802 brightness=896
1920x1200 t=4.089457 latency=2.165979 brightness=909
1920x1200 t=4.205064 latency=2.161043 brightness=836
1920x1200 t=4.317453 latency=2.180339 brightness=908
1920x1200 t=4.433037 latency=2.172433 brightness=945
1920x1200 t=4.541008 latency=2.167379 brightness=922
1920x1200 t=4.655401 latency=2.155513 brightness=956
1920x1200 t=4.770161 latency=2.173663 brightness=843
1920x1200 t=4.885308 latency=2.168308 brightness=934
1920x1200 t=5.000802 latency=2.163257 brightness=840
1920x1200 t=5.115584 latency=2.158481 brightness=937
1920x1200 t=5.231182 latency=2.177084 brightness=883
1920x1200 t=5.346843 latency=2.172450 brightness=902
1920x1200 t=5.462503 latency=2.167889 brightness=865
1920x1200 t=5.573996 latency=2.163009 brightness=854
1920x1200 t=5.708317 latency=2.176815 brightness=899
```

In the about output, the latency increases from very little to more than two seconds (note that the time t is ticking normally). This is unexpected since we only wait 100ms and the exposure time is very small. We should be getting fresh frames.

Any ideas how we can fix this? Thanks in advance!

PS: I'm using USB3 cameras.

The script:

```
import time
import numpy as np
from pypylon import pylon
from pypylon import genicam

countOfImagesToGrab = 50

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

# Use minimal buffering so we get fresh video frames, hopefully.
camera.StreamGrabber.MaxNumBuffer.SetValue(1)
camera.MaxNumBuffer.SetValue(1) 

print("Using device ", camera.GetDeviceInfo().GetModelName())

t0 = time.clock_gettime(time.CLOCK_MONOTONIC_RAW)

def get_clock_difference():
    global clock_difference
    l = []
    for i in range(0, 100):
        camera.TimestampLatch.Execute()
        raw_time = time.clock_gettime(time.CLOCK_MONOTONIC_RAW)
        camera_time = camera.TimestampLatchValue.GetValue()*1e-9
        clock_difference = raw_time-camera_time
        l.append(clock_difference)
    l = np.array(l)
    clock_difference = np.median(l)
    return clock_difference
clock_difference = get_clock_difference()

camera.StartGrabbingMax(countOfImagesToGrab, pylon.GrabStrategy_LatestImageOnly) # Doesn't work.
#camera.StartGrabbingMax(countOfImagesToGrab, pylon.GrabStrategy_UpcomingImage) # Not supported for USB cameras.
#camera.StartGrabbingMax(countOfImagesToGrab, pylon.GrabStrategy_LatestImages) # Doesn't work either.
#camera.StartGrabbingMax(countOfImagesToGrab, pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByUser) # Doesn't work either.

while camera.IsGrabbing():

    frame = None

    if True:
        # Try to clear the buffer, but this doesn't seem to work.
        while True:
            tmp_frame = camera.RetrieveResult(0, pylon.TimeoutHandling_Return)
            if not tmp_frame: break
            if not frame is None:
                print("DISCARDING SINGLE FRAME")
            frame = tmp_frame

    if not frame:
        frame = camera.RetrieveResult(int(5000), pylon.TimeoutHandling_ThrowException)

    if frame.GrabSucceeded():
        # Access the image data.
        dt = time.clock_gettime(time.CLOCK_MONOTONIC_RAW) - (frame.TimeStamp*1e-9 + clock_difference)
        brightness = np.sum(frame.Array)
        print("%dx%d"%(frame.Width, frame.Height), "t=%.06f"%(time.clock_gettime(time.CLOCK_MONOTONIC_RAW)-t0), "latency=%.06f"%dt, "brightness=%d"%brightness)
    else:
        print("Error:", frame.ErrorCode, frame.ErrorDescription)

    time.sleep(.1) # Note: we're simulating here the case where the system is doing some heavy processing and can't keep up with frames.
    frame.Release()

camera.Close()
```

SMA2016a, 2022-11-23:
try camera.StartGrabbingMax( pylon.GrabStrategy_LatestImages)   and do not cloce and open the camera all the time. It is better that you use the software trigger therefor keep the camera open.  This approach will save freeing and reallocation memory

keesjan123, 2022-11-23:
Hi, thanks for your quick response. I already tried GrabStrategy_LatestImages (see the comment in the script), and also the camera is opened only once in the beginning and closed once at the end. Unfortunately, none of the grab strategy constants seem to have an effect on the problem.

keesjan123, 2022-11-29:
Ok, I rewrote the above program in C++ and it looks like the problem also exists in Pylon's C++ API.

See below (based on the Grab.cpp example from pylon, and the Makefile can be taken from Grab.cpp as well):

```
#include <pylon/PylonIncludes.h>
#include <pylon/BaslerUniversalInstantCamera.h>
#include <chrono>
#include <unistd.h>
#include <algorithm>

using namespace Pylon;
using namespace std;

static const uint32_t c_countOfImagesToGrab = 50;

auto program_start_time = std::chrono::steady_clock::now();

inline
double
get_time(void)
{
    return double(std::chrono::duration_cast<std::chrono::nanoseconds>(std::chrono::steady_clock::now() - program_start_time).count())*1e-9;
}

static
double
get_clock_difference(Pylon::CBaslerUniversalInstantCamera &camera)
{
    const int N = 101;

    std::vector<double> l;

    for(int i = 0; i < N; ++i)
    {
        camera.TimestampLatch.Execute();
        double raw_time = get_time();
        double camera_time = double(camera.TimestampLatchValue.GetValue()) * 1e-9;
        double clock_difference = raw_time-camera_time;
        l.push_back(clock_difference);
    }

    // Simple way to compute median.
    sort(l.begin(), l.end());
    double clock_difference = l[int(N/2)];

    return clock_difference;
}

int main( int /*argc*/, char* /*argv*/[] )
{
    int exit_code = 0;

    Pylon::PylonAutoInitTerm autoInitTerm;

    try
    {
        Pylon::CBaslerUniversalInstantCamera camera( CTlFactory::GetInstance().CreateFirstDevice() );
        // Pylon::CInstantCamera camera( CTlFactory::GetInstance().CreateFirstDevice() );

        cout << "Using device " << camera.GetDeviceInfo().GetModelName() << endl;

        // The parameter MaxNumBuffer can be used to control the count of buffers
        // allocated for grabbing. The default value of this parameter is 10.
        camera.MaxNumBuffer = 1;

        // Start the grabbing of c_countOfImagesToGrab images.
        // The camera device is parameterized with a default configuration which
        // sets up free-running continuous acquisition.
        camera.StartGrabbing( c_countOfImagesToGrab );

        double clock_difference = get_clock_difference(camera);

        // This smart pointer will receive the grab result data.
        CGrabResultPtr ptrGrabResult;

        // Camera.StopGrabbing() is called automatically by the RetrieveResult() method
        // when c_countOfImagesToGrab images have been retrieved.
        while (camera.IsGrabbing())
        {
            // Wait for an image and then retrieve it. A timeout of 5000 ms is used.
            camera.RetrieveResult( 5000, ptrGrabResult, TimeoutHandling_ThrowException );

            // Image grabbed successfully?
            if (ptrGrabResult->GrabSucceeded())
            {
                double raw_time = get_time();
                double camera_time = ptrGrabResult->GetTimeStamp()*1e-9;
                double dt = raw_time - (camera_time + clock_difference);

                cout << ptrGrabResult->GetWidth() << "x" << ptrGrabResult->GetHeight() << " latency=" << dt << endl;
            }
            else
            {
                cout << "Error: " << std::hex << ptrGrabResult->GetErrorCode() << std::dec << " " << ptrGrabResult->GetErrorDescription() << endl;
            }

            ::usleep(100000); // Note: we're simulating here the case where the system is doing some heavy processing and can't keep up with frames.
        }
    }
    catch (const GenericException& e)
    {
        // Error handling.
        cerr << "An exception occurred." << endl
            << e.GetDescription() << endl;
        exit_code = 1;
    }

    return exit_code;
}
```

In the above code, the latency increases to about 2 seconds (!) while we're only sleeping 100ms (to simulate a heavy workload) after every frame.

While this makes this not a Python-only problem, I'm still hoping that anyone here can help me and tell me what I'm doing wrong, or if there really is a bug in Pylon.

thiesmoeller, 2022-11-29:
From what you describe I would say that your USB transfer rate is too low. 

The camera assumes that it can transfer data at full USB 3.0 but your hw is limited to a much lower data rate.

This will queue up video data in the device memory and will show the effect you described.

The buffer handling parameters OneByOne, Latest .. only control the buffers _after_ have been received to the host.

... e.g TI XHCIs can only transfer 280MB/s ..
The parameter to limit the camera USB bandwidth is DeviceLinkThroughputLimit.

Another reason could be is that you have raised a user thread priority way above pylon's  internal grab and thus we can't transfer the data fast enough to internal receive buffers

SMA2016a, 2022-11-30:
@keesjan123   Let us do small meeting later to understand your use case and find out a good working solution for you. I call you later.

keesjan123, 2022-11-30:
Ok, I had a discussion with people from Basler, and it turns out that the camera-hardware also has a queue, which cannot be trivially circumvented from within Pylon by using the grab strategies. The recommended solution is to avoid streaming, and instead perform a software trigger to start acquisition of a each new frame, in a loop to simulate streaming. This way, reading of frames and emptying of buffers happens in lockstep.

This issue can be marked closed.

closed 2022-11-30 by keesjan123