basler/pypylon #560: libpylonbase.so.7.2: cannot open shared object file: No such file or directory

AarathyR, 2022-11-24:
**I am using Basler camera and wanted it to work from the python code. I downloaded pylon_master from github and installed it by using setup.py in it. My code is** 

from pypylon import pylon

tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices()
for device in devices:
    print(device.GetFriendlyName())

**But I got error like**

Traceback (most recent call last):
  File "1.py", line 1, in <module>
    from pypylon import pylon
  File "/usr/local/lib/python3.8/dist-packages/pypylon-2022.11.23.dev66612+pylon7.2.1-py3.8-linux-x86_64.egg/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: libpylonbase.so.7.2: cannot open shared object file: No such file or directory

Can someone guide me whats wrong happening here.

SMA2016a, 2022-11-25:
why do you not install using pip? maybe this will fix the issue?

It seems to be the environment variable PYLON_ROOT is missing.

thiesmoeller, 2022-11-25:
pypylon is not yet enabled to work with pylon 7.x

There is no difference to use pylon 6.x as base. This is what we use if you install via pip install

closed 2023-04-26 by AarathyR