basler/pypylon #561: where I can add the pylon-config file in windows 11

sulaimansiddiqi233, 2022-11-26:
hello 
@SMA2016a @m-binev @thiesmoeller
we just want to run a make file in windows 11 but the get the errors as you can see in attachment file. after that Basler team send as a pylon-config file and they told to put it in the path (/opt/pylon/bin) but unfortunately in window 11 we could not find the path . so could help us in the issue. 
[image]

thank you.

SMA2016a, 2022-11-26:
pylon sdk can not easy builld with mingw.  You need to tweak the header files.  pylon-config is just for Linux user only.

sulaimansiddiqi233, 2022-11-28:
@SMA2016a so how can I run the makefile in windows.

sulaimansiddiqi233, 2024-03-06:
we solve it

closed 2024-03-06 by sulaimansiddiqi233