basler/pypylon #564: Export Feature Tree

KohChr, 2022-12-06:
I would like to export the entire Featre Tree (all categories, features and values) of the Basler camera e.g. into a .json file to be able to trace the state of the camera in case of an error.
Is there a simple way to query the feature tree via pypylon?

thiesmoeller, 2022-12-06:
See this sample code here https://github.com/basler/pypylon/issues/533#issuecomment-1240491053

KohChr, 2022-12-07:
I would like to do exactly in pyhton what is possible in pylon Viwer under Copy Camera Features.
Writing a parser for all entries in the device tree is not a nice solution.

[image]

thiesmoeller, 2022-12-07:
This function from the viewer is available in python see here https://github.com/basler/pypylon/blob/2f787d41cf9a8140382a0a4a317d16747e9289c1/samples/parametrizecameraloadandsaveconfig.py

The difference is: in the PFS file we only store the user settable values. Your ticket asked for the _entire_ tree. Which would also include read-only features like device temperature

KohChr, 2022-12-07:
Thank you for the prompt feedback. 
There is still a mismatch between the parameter names. For example, if the exposure time is set in the pylon viewer or via pypylon.pylon.InstantCamera, the **ExposureTime** parameter must be used. 
In the example program parametrizecameraloadandsaveconfig.py the parameter for the exposure time is named **ExposureTimeRaw** (like in the GigE cameras)
How can this difference in the name of the parameter be explained?

thiesmoeller, 2022-12-07:
The persistance mechanism stores, depending on the camera model , the internal integer representations of floating point features to precisely restore the persisted value.

If ypu really want to persist exactly the  values you see as user, tou would have to use the manual method described above. 
It has to be extended a bit to also unroll selectors ....