basler/pypylon #566: change camera parameters

imtamo, 2022-12-09:
I am a beginner. I am trying to capture one image at a time with slightly different camera exposure times and gains. But I am not sure if it is working well.

- Basler GigE acA1300-60gmNIR

`
   

    from pypylon import pylon
    import matplotlib.pyplot as plt
    from tqdm import tqdm 
    expo_times = range(10, 1000 + 1, 10)
    gains = range(0, 3 + 1, 1)
    
    tl_factory = pylon.TlFactory.GetInstance()
    camera = pylon.InstantCamera()
    camera.Attach(tl_factory.CreateFirstDevice())
    camera.Open()
    for gain in tqdm(gains):
        camera.GainRaw.SetValue(gain)
                
        for expo_time in expo_times:
            camera.ExposureTimeRaw.SetValue(expo_time)
            camera.Close()
            camera.Open()
            camera.StartGrabbing(1)
            grab = camera.RetrieveResult(2000, pylon.TimeoutHandling_Return)
            print(f'-->Gain：{camera.GainRaw.Value}')
            print(f'-->Exposure Time：{camera.ExposureTimeRaw.Value}')
                    
            if grab.GrabSucceeded():
                img_raw = grab.GetArray()
                plt.imshow(img_raw, cmap='gray')
                plt.show()
`

thiesmoeller, 2022-12-11:
Welcome :-)

when you are new to pypylon I highly recommend to check the companion repository [pypylon-samples](https://github.com/basler/pypylon-samples). There we reference an hour long webinar and have a number of samples and notebooks.

To your task:

What should your program do in the end?

imtamo, 2022-12-12:
Thanks for the reply.

My program aims to take one image for each of the gains 0, 1, 2, and 3 when the exposure time is varied from 10 to 1000 in 10 increments.
I would like to check if the above program achieves this objective.

thiesmoeller, 2022-12-13:
as calling matplotlib `plt.show()` would only make sense in e.g. a jupyterlab environment.. ( you would have to force close the window to move to next step )
here you can see for reference an example to do a parameter sweep with an opencv windows ( see also in example folder )

```python
from pypylon import pylon as py
from tqdm import tqdm
import numpy as np
import cv2

# transport layer factory
tlf = py.TlFactory.GetInstance()
# get first device
cam = py.InstantCamera(tlf.CreateFirstDevice())
# open
cam.Open()

# depending on your camera model ace1 GigE vs newer cameras
# the features can differ GainRaw/GainAbs vs. Gain
# see docs.baslerweb.com for details
cam_gain = cam.GainRaw
cam_exptime = cam.ExposureTimeAbs

# example to
# 
gain_range = (0,1,2,3)
exptime_range = range(10, 1000+1, 10)

# the fastest procedure to get an
# image with _all_ previously set features applied
# is to use SoftwareTrigger on the camera
cam.TriggerSelector = "FrameStart"
cam.TriggerSource = "Software"
cam.TriggerMode = "On"

# try to enable a monochrome pixel format
cam.PixelFormat = "Mono8"
# disable auto exposure if set
cam.ExposureAuto = "Off"

cam.StartGrabbing()

for gain in tqdm(gain_range, desc="Gain", ncols=80):
    # return from outer loop
    if not cam.IsGrabbing():
        break
    cam_gain.SetValue(gain)
    for expo_time in tqdm(exptime_range, desc="Expo", ncols=80, leave=False):
        # return from inner loop
        if not cam.IsGrabbing():
            break

        # return from inner loop
        cam_exptime.SetValue(expo_time)

        # execute Software trigger to get
        # an image with above gain and exposure time
        # query the camera if ready to accept new trigger
        if cam.WaitForFrameTriggerReady(200, py.TimeoutHandling_ThrowException):
           cam.TriggerSoftware.Execute()

        # use context manager to let pypylon take care of release the grab result
        # after usage wait for the image to return
        # timeout has to be at least longest exposure time in the array
        with cam.RetrieveResult(2000) as grab_result:
            if grab_result.GrabSucceeded():
                img = grab_result.GetArray()
                # use cv2 to display a live video image
                cv2.namedWindow("title", cv2.WINDOW_NORMAL)
                cv2.imshow("title", img)
                k = cv2.waitKey(1)
                if k == 27:
                    cam.StopGrabbing()
                    break
            else:
                print("GrabFailed")

cam.StopGrabbing()
cam.Close()

cv2.destroyAllWindows()

```

imtamo, 2022-12-14:
Thank you.
I'll try that.