basler/pypylon #568: No grab result data is referenced. Cannot access NULL pointer

Lakshmantlrao, 2022-12-12:
Hey,

Using two acA2040-55uc cameras that are usb on Ubuntu 16.04 running on Intel i7 8th gen. Have put the cams at Max fps of 23 fps and using a software trigger for frames. Could see the error in the title occuring simultaneously in both cameras at random. Any information would be of great help ...Thanks in advance!!!

thiesmoeller, 2022-12-12:
please post a piece of sample code, that triggers this error

Lakshmantlrao, 2022-12-13:
Hey,

Thank You for your response. Below is a code snippet we are using  which has be referenced from the samples in the repo. We are using pypylon 1.8.0 on python3.6.10. We are getting exception in the code blocks 3 and 4 at the same time

Exception printed on console-
camera1------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)
camera2------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)

                        try:                                  # code block 1
                            camera1.ExecuteSoftwareTrigger()
                           
                        except Exception as e:
                            print(str(e))
                            
                       
                        try:                                      # code block 2
                            camera2.ExecuteSoftwareTrigger()
                            
                        except Exception as e:
                            print(str(e))

                     try:                                 # code block 3
                        if grabResult1.GrabSucceeded():
                            img1 = grabResult1.GetArray()
                            grabResult1.Release()
                     except Exception as e:
                             print("camera1------ ",str(e))

                     try:                                     # code block 4 
                        if grabResult2.GrabSucceeded():
                            img2 = grabResult2.GetArray()
                            grabResult2.Release()
                     except Exception as e:
                            print("camera2------ ",str(e))

thiesmoeller, 2022-12-13:
the interesting part is missing.... 
Where are your RetrieveResult calls in relation to what you showed last...

The error message indicates, that either you or python's garbage collector has already remove the GrabResult ( result of calling RetrieveResult ) 
when you try to access the element ( e.g. in grabResult1.GrabSucceeded() )

Lakshmantlrao, 2022-12-13:
The retrieveresult command for both the cameras are placed just above the GrabSucceeded condition for both the cameras just as shown

                    try:
                        grabResult2 = camera2.RetrieveResult(500, pylon.TimeoutHandling_Return)
                    except Exception as e:
                        print("Exception in Camera2 Grab Retrive",e)
                    
              
             
               try:
                    if grabResult2.GrabSucceeded():
                            img2 = grabResult2.GetArray()
                            grabResult2.Release()

thiesmoeller, 2022-12-13:
grabresult2 is invalid when you got the exception above.

Lakshmantlrao, 2022-12-13:
here the  Exception getting block2   last line  .
try:
      grabResult2 = camera2.RetrieveResult(500, pylon.TimeoutHandling_Return)
                    except Exception as e:
                        print("Exception in Camera2 Grab Retrive",e)
                        logging.debug("Second Cam frame grab failure-"+str(datetime.now()))
                        #raise Exception("Cameras Camera2 Grab Retrive "+str(e))
                    cam2_grab_time=int((time.time()-cam2_grab_time)*1000)
                    cam2_grab_convert_and_save=time.time()
                    try:
                        if grabResult2.GrabSucceeded():
                            img2 = grabResult2.GetArray()
                            grabResult2.Release()
                            if Recording_Status=="Start_Survey":
                                try:
                                    q_cam_save2.put_nowait((img2,scene_record_path2,save_file_time+"_Distress2.jpg"))
                                    ctr1+=1
                                except:
                                    try:
                                        temp_cam_save2.put_nowait((img2,scene_record_path4,save_file_time+"_Distress2.jpg"))
                                    except:
                                        cam2_status="Camera_Frame_Save_Missing"
                                        save_miss_counter_cam2+=1
                                        cam2_save_status=False
                                        print(save_file_time,chainage,"Save Camera2 frame Missed:",save_miss_counter_cam2)
                                        logging.debug(str([save_file_time,chainage,"Save Camera2 frame Missed:",save_miss_counter_cam2]))
                                        pass
                        else:
                            print("Grabresult2 Failed",save_file_time,chainage)
                            grab_failed_cam2+=1
                            cam2_grab_status=False
                            print("Error2: ", grabResult2.ErrorCode, grabResult2.ErrorDescription)
                            logging.debug(str(["Error in 2",grabResult2.ErrorCode, grabResult2.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        print("camera2",e)  ###########<<<<**here  i getting error>

thiesmoeller, 2022-12-14:
can you please format your code using
````
```python
<your code>
```
````

as it is not possible to see the indentation of your code, which is really critical.
e.g. when you call `grabResult.Release()`

Lakshmantlrao, 2023-01-26:
`                    try:
##                        if camera1.GetGrabResultWaitObject().Wait(75):
                        
                        grabResult1 = camera1.RetrieveResult(500, pylon.TimeoutHandling_Return)
                        
                    except Exception as e:
                        
                        print("Exception in Camera1 Grab Retrive",e)
                        logging.debug("First frame grab fail-"+str(datetime.now()))
                        #raise Exception("Cameras Camera1 Grab Retrive "+str(e))
                    cam1_grab_time=int((time.time()-cam1_grab_time)*1000)
                    
                    cam1_grab_convert_and_save=time.time()
                    try:
                        if grabResult1.GrabSucceeded():
                            img1 = grabResult1.GetArray()
                            grabResult1.Release()
                            if Recording_Status=="Start_Survey":
                                try:
                                    q_cam_save1.put_nowait((img1,scene_record_path1,save_file_time+"_Distress1.jpg"))
                                except:
                                    try:
                                        temp_cam_save1.put_nowait((img1,scene_record_path3,save_file_time+"_Distress1.jpg"))
                                    except:
                                        save_miss_counter_cam1+=1
                                        cam1_save_status=False
                                        cam1_status="Camera_Frame_Save_Missing"
                                        print(save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1)
                                        logging.debug(str([save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1]))
                                        pass
##                                save1=time.time()-save1
##                                cam1_grab_convert_and_save=int((cam1_grab_convert_and_save+save1)*1000)
    ##                            q_send_gui.put_nowait((img1,save_file_time,"2","Camera_Ok"))
                            
                        else:
                            print("Grabresult1 Failed",save_file_time,chainage)
                            grab_failed_cam1+=1
                            cam1_grab_status=False
                            print("Error1: ", grabResult1.ErrorCode, grabResult1.ErrorDescription)
                            logging.debug(str(["Error in 1",grabResult1.ErrorCode, grabResult1.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        try:
                            try:
                                grabResult1.Release()
                            except:
                                pass
                            if time.time()-restart_t1>5:
                                restart_t1=time.time()
                                camera1=first_cam_initialise()#first_cam_reinit()
                                
                        except:
                            print(save_file_time,chainage,"Exception in Camera1 grabResult1.GrabSucceeded() :",e)
                            grab_failed_cam1+=1
                            cam1_grab_status=False
                            logging.debug([save_file_time," Error1 in Camera1 grabResult1.GrabSucceeded() :",e])
                            pass
                        
                    cam1_grab_convert_and_save=int((time.time()-cam1_grab_convert_and_save)*1000)
                    cam2_grab_time=time.time()
                    try:
                        grabResult2 = camera2.RetrieveResult(500, pylon.TimeoutHandling_Return)
                    except Exception as e:
                        print("Exception in Camera2 Grab Retrive",e)
                        logging.debug("Second Cam frame grab failure-"+str(datetime.now()))
                        #raise Exception("Cameras Camera2 Grab Retrive "+str(e))
                    cam2_grab_time=int((time.time()-cam2_grab_time)*1000)
                    cam2_grab_convert_and_save=time.time()
                    try:
                        if grabResult2.GrabSucceeded():
                            img2 = grabResult2.GetArray()
[... 14 lines cut ...]
                                        pass
                        else:
                            print("Grabresult2 Failed",save_file_time,chainage)
                            grab_failed_cam2+=1
                            cam2_grab_status=False
                            print("Error2: ", grabResult2.ErrorCode, grabResult2.ErrorDescription)
                            logging.debug(str(["Error in 2",grabResult2.ErrorCode, grabResult2.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        try:
                            try:
                                grabResult2.Release()
                            except:
                                pass
                            if time.time()-restart_t2>5:
                                restart_t2=time.time()
                                camera2=second_cam_initialise()#second_cam_reinit()
                                
                        except:
                            print(save_file_time,chainage,"Exception in Camera2 grabResult2.GrabSucceeded() :",e)
                            grab_failed_cam2+=1
                            cam2_grab_status=False
                            logging.debug([save_file_time," Error2 in Camera2 grabResult2.GrabSucceeded() :",e])
                            pass
                    cam2_grab_convert_and_save=int((time.time()-cam2_grab_convert_and_save)*1000)
                    `

Lakshmantlrao, 2023-01-26:
```python
<                    try:

                        
                        grabResult1 = camera1.RetrieveResult(500, pylon.TimeoutHandling_Return)
                        
                    except Exception as e:
                        
                        print("Exception in Camera1 Grab Retrive",e)
                        logging.debug("First frame grab fail-"+str(datetime.now()))
                        #raise Exception("Cameras Camera1 Grab Retrive "+str(e))
                    cam1_grab_time=int((time.time()-cam1_grab_time)*1000)
                    
                    cam1_grab_convert_and_save=time.time()
                    try:
                        if grabResult1.GrabSucceeded():
                            img1 = grabResult1.GetArray()
                            grabResult1.Release()
                            if Recording_Status=="Start_Survey":
                                try:
                                    q_cam_save1.put_nowait((img1,scene_record_path1,save_file_time+"_Distress1.jpg"))
                                except:
                                    try:
                                        temp_cam_save1.put_nowait((img1,scene_record_path3,save_file_time+"_Distress1.jpg"))
                                    except:
                                        save_miss_counter_cam1+=1
                                        cam1_save_status=False
                                        cam1_status="Camera_Frame_Save_Missing"
                                        print(save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1)
                                        logging.debug(str([save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1]))
                                        pass
##                                save1=time.time()-save1
##                                cam1_grab_convert_and_save=int((cam1_grab_convert_and_save+save1)*1000)
    ##                            q_send_gui.put_nowait((img1,save_file_time,"2","Camera_Ok"))
                            
                        else:
                            print("Grabresult1 Failed",save_file_time,chainage)
                            grab_failed_cam1+=1
                            cam1_grab_status=False
                            print("Error1: ", grabResult1.ErrorCode, grabResult1.ErrorDescription)
                            logging.debug(str(["Error in 1",grabResult1.ErrorCode, grabResult1.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        try:
                            try:
                                grabResult1.Release()
                            except:
                                pass
                            if time.time()-restart_t1>5:
                                restart_t1=time.time()
                                camera1=first_cam_initialise()#first_cam_reinit()
                                
                        except:
                            print(save_file_time,chainage,"Exception in Camera1 grabResult1.GrabSucceeded() :",e)
                            grab_failed_cam1+=1
                            cam1_grab_status=False
                            logging.debug([save_file_time," Error1 in Camera1 grabResult1.GrabSucceeded() :",e])
                            pass
                        
                    cam1_grab_convert_and_save=int((time.time()-cam1_grab_convert_and_save)*1000)
                    cam2_grab_time=time.time()
                    try:
                        grabResult2 = camera2.RetrieveResult(500, pylon.TimeoutHandling_Return)
                    except Exception as e:
                        print("Exception in Camera2 Grab Retrive",e)
                        logging.debug("Second Cam frame grab failure-"+str(datetime.now()))
                        #raise Exception("Cameras Camera2 Grab Retrive "+str(e))
                    cam2_grab_time=int((time.time()-cam2_grab_time)*1000)
                    cam2_grab_convert_and_save=time.time()
                    try:
                        if grabResult2.GrabSucceeded():
[... 16 lines cut ...]
                        else:
                            print("Grabresult2 Failed",save_file_time,chainage)
                            grab_failed_cam2+=1
                            cam2_grab_status=False
                            print("Error2: ", grabResult2.ErrorCode, grabResult2.ErrorDescription)
                            logging.debug(str(["Error in 2",grabResult2.ErrorCode, grabResult2.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        try:
                            try:
                                grabResult2.Release()
                            except:
                                pass
                            if time.time()-restart_t2>5:
                                restart_t2=time.time()
                                camera2=second_cam_initialise()#second_cam_reinit()
                                
                        except:
                            print(save_file_time,chainage,"Exception in Camera2 grabResult2.GrabSucceeded() :",e)
                            grab_failed_cam2+=1
                            cam2_grab_status=False
                            logging.debug([save_file_time," Error2 in Camera2 grabResult2.GrabSucceeded() :",e])
                            pass
                    cam2_grab_convert_and_save=int((time.time()-cam2_grab_convert_and_save)*1000)
                    >
```

Lakshmantlrao, 2023-01-26:
Hey,

Thank You for your response. Below is a code snippet we are using which has be referenced from the samples in the repo. We are using pypylon 1.8.0 on python3.8 We are getting exception in the code blocks 3 and 4 at the same time

i added my related code above comment

Exception printed on console-
camera1------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)
camera2------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)

Lakshmantlrao, 2023-01-31:
Exception printed on console-
camera1------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)
camera2------ No grab result data is referenced. Cannot access NULL pointer. : RuntimeException thrown (file 'GrabResultPtr.cpp', line 87)

```python
<                  try:
                        
                        grabResult1 = camera1.RetrieveResult(500, pylon.TimeoutHandling_Return)    ###### Gets frame buffer from camera
                        
                    except Exception as e:
                        
                        print("Exception in Camera1 Grab Retrive",e)
                        logging.debug("First frame grab fail-"+str(datetime.now()))
                        if camera1.IsOpen():
                            camera1.StopGrabbing()
                            camera1.Close()
                            camera1=second_cam_initialise()
                    cam1_grab_time=int((time.time()-cam1_grab_time)*1000)
                    
                    cam1_grab_convert_and_save=time.time()
                    try:
                        if grabResult1.GrabSucceeded():          #<<<<<<<<<<<<<<<<<<<here getting error
                            img1 = grabResult1.GetArray()           
                            grabResult1.Release()
                            if Recording_Status=="Start_Survey":
                                try:
                                    q_cam_save1.put_nowait((img1,scene_record_path1,save_file_time+"_Distress1.jpg"))
                                except:
                                    try:
                                        temp_cam_save1.put_nowait((img1,scene_record_path3,save_file_time+"_Distress1.jpg"))
                                    except:
                                        save_miss_counter_cam1+=1
                                        cam1_save_status=False
                                        cam1_status="Camera_Frame_Save_Missing"
                                        print(save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1)
                                        logging.debug(str([save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1]))
                                        pass
                            
                        else:
                            print("Grabresult1 Failed",save_file_time,chainage)
                            grab_failed_cam1+=1
                            cam1_grab_status=False
                            print("Error1: ", grabResult1.ErrorCode, grabResult1.ErrorDescription)
                            logging.debug(str(["Error in 1",grabResult1.ErrorCode, grabResult1.ErrorDescription,time.ctime()]))
                    except Exception as e:
                        print("Camera1----",str(e))
                        >
```

thiesmoeller, 2023-01-31:
You are not releasing the grab results in all error paths. 

To better handle this, a context manager exists:

```python

with cam.RetrieveResult(1000) as res:
      ... do something with the result ...
```

The context manager will take care of releasing the result on any path out of the context

As a side note: ... your code has a really high complexity due to the nested exception handlers.  You might refactor this to make your main intent of the code more visible

Lakshmantlrao, 2023-02-01:
hey ,
thanks  for your feedback,
i  try  with some changes in my code but i getting  this issue is shown below  :

""
Exception in Camera1 Grab Retrive::: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1037)

""
this is my code:
```python
<                 try:
                        grabResult1=camera1.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
                        cam1_grab_time=int((time.time()-cam1_grab_time)*1000)
                        cam1_grab_convert_and_save=time.time()
                        try:
                            if grabResult1.GrabSucceeded():              ####### if frame buffer is successfully received from camera 
                                img1 = grabResult1.GetArray()            ####### gets array from frame buffer
                                if Recording_Status=="Start_Survey":
                                    try:
                                        q_cam_save1.put_nowait((img1,scene_record_path1,save_file_time+"_Distress1.jpg"))
                                    except:
                                        try:
                                            temp_cam_save1.put_nowait((img1,scene_record_path3,save_file_time+"_Distress1.jpg"))
                                        except:
                                            save_miss_counter_cam1+=1
                                            cam1_save_status=False
                                            cam1_status="Camera_Frame_Save_Missing"
                                            print(save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1)
                                            logging.debug(str([save_file_time,chainage,"Save Camera1 frame Missed:",save_miss_counter_cam1]))
                                            pass
                                
                            else:
                                print("Grabresult1 Failed",save_file_time,chainage)
                                grab_failed_cam1+=1
                                cam1_grab_status=False
                                print("Error1: ", grabResult1.ErrorCode, grabResult1.ErrorDescription)
                                logging.debug(str(["Error in 1",grabResult1.ErrorCode, grabResult1.ErrorDescription,time.ctime()]))
                            grabResult1.Release() ### i added
                        except Exception as e:
                            print("Camera1 ..grab not succed----",str(e))
                            grabResult1.Release() ### i added
                        
                    except Exception as e:
                        
                        print("Exception in Camera1 Grab Retrive:::",e)
                       # logging.debug("First frame grab fail-"+str(datetime.now()))
                        grabResult1.Release()>
```

thiesmoeller, 2023-02-01:
Is any of the hints in the exception valid for you?

To get this running create a second reduced version where you just grab the images .. and then step by step add your processing back