basler/pypylon #570: Not able to detect the usb3 camera after PC power unplug

maxwxzheng, 2022-12-13:
If the PC is unplugged from power while the PC and the camera are running, we can't detect the camera after restarting the PC. We have to unplug and replug the camera from the PC to recover from this. Is there a better way to handle this situation?

We are working with Ubuntu 20.04. Basler acA2440-75uc.

SMA2016a, 2022-12-14:
This issue,  does not seem to  related to pypylon.   I think this could be USB Lib or USB Host controller. Please check with your local  distributor to check if your camera has got latest firmware. Then let us know which USB host controller and what kind of PC you are using?

maxwxzheng, 2022-12-14:
I reached out to Basler US support team. I'll close this.

closed 2022-12-14 by maxwxzheng