basler/pypylon #571: Unable to grab images from Linux container on Windows

tc360950, 2022-12-14:
Hi, 
I have Basler acA2440-20gc camera connected to Windows host through the network. 
The camera is visible in pylon Viewer and works properly. 

However, my goal is to be able to use the camera from inside Linux docker container. 
I run the container with -P flag and use IP shown by pylon Viewer to connect to the camera. 
I also set stream grabber port to a fixed value. 

The connection seems to be successful - I am able to retrieve camera parameters and update them.
Alas, when I try to grab frames from the camera I get a timeout:
`
Traceback (most recent call last):
  File "/app/viewer.py", line 33, in <module>
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  File "/usr/local/lib/python3.10/site-packages/pypylon/pylon.py", line 3511, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1037
`

The issue occurs both on WSL2 and Hyper-V based backends for Docker Desktop. 

I use the following code to connect to the camera: 
`
from pypylon import pylon

ip_address = '169.254.80.50'
tlf = pylon.TlFactory.GetInstance()
ptl = tlf.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = tlf.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)
camera.Open()
camera.StreamGrabber.DestinationPort = 50100;

# demonstrate some feature access
camera.Width.SetValue(camera.Width.GetMin())
camera.Height.SetValue(camera.Height.GetMin())

# Up to this point everything goes flawlessly 
numberOfImagesToGrab = 2
camera.StartGrabbingMax(numberOfImagesToGrab)

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data.
        print("SizeX: ", grabResult.Width)
        print("SizeY: ", grabResult.Height)
        img = grabResult.Array
        print("Gray value of first pixel: ", img[0, 0])

    grabResult.Release()
camera.Close()
 `

(This script works just fine when run directly on my Windows host).

Any idea why this error occurs? Is there some guide available for connecting from Docker on Windows?

Thanks for the help.

SMA2016a, 2022-12-14:
Try to use  like this without passing any parameter. 

ptl = tlf.CreateTl()

tc360950, 2022-12-15:
Thanks for the answer!
 I got error 
``` TypeError: Wrong number or type of arguments for overloaded function 'TlFactory_CreateTl'. ```

This is confirmed by SDK docs: 

https://ja.docs.baslerweb.com/pylonapi/cpp/class_pylon_1_1_c_tl_factory

It seems it's impossible to call it without any parameters.

thiesmoeller, 2022-12-15:
Your ace model can only stream into same network as your application is in.

Is this the case in your environment ?

tc360950, 2022-12-15:
It is, I can receive frames from the host without any problem. But it's not the case for my Docker container which by default is in docker network. As far as I know, it's impossible to place docker containers directly on the host network in Windows - am I right saying that it means I will not be able to use the camera from inside docker?

tc360950, 2022-12-15:
@thiesmoeller Could you please point me to some online documentation for this? I was not able to find anything here 
https://docs.baslerweb.com/aca2440-20gc 

Thank you.

SMA2016a, 2022-12-15:
@tc360950   try this code

 ip_address = '169.254.80.50'

    tlf = pylon.TlFactory.GetInstance()

    #ptl = tlf.CreateTl('BaslerGigE')
    ptl = pylon.TlFactory.GetInstance()
    

    empty_camera_info = pylon.DeviceInfo()

    empty_camera_info.SetPropertyValue('IpAddress', ip_address)

    camera_device = tlf.CreateDevice(empty_camera_info)

    camera = pylon.InstantCamera(camera_device)

tc360950, 2022-12-15:
@SMA2016a 
Thanks for the response I really appreciate your assistance. Now I can't event connect to he camera
```
File "/usr/local/lib/python3.10/site-packages/pypylon/pylon.py", line 1613, in CreateDevice
    return _pylon.TlFactory_CreateDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
```

Is there any rationale to this change? Maybe this will help me to tinker on my own.

tc360950, 2022-12-16:
@thiesmoeller 
Our cam is connected with RJ-45 network cable straight into AdLink PCIe-GIE74P on the host

thiesmoeller, 2022-12-16:
Have you worked through the options in https://techcommunity.microsoft.com/t5/itops-talk-blog/docker-host-network-alternatives-for-windows-containers/ba-p/3390115 ?

You have to find a setup where the camera and the application is in the same subnet.
Anyhow performance will degrade if docker has to software route 100MB/s from the real network interface to your application.

If this all is not working for you. Isn't switching the host to Linux too an option ?

thiesmoeller, 2022-12-16:
It's not explicitly documented but in https://docs.baslerweb.com/network-configuration-(gige-cameras) you see the possible network architectures.

Your setup has a SW router in docker above your network card.

As I'm not using this Linux on Win docker setup.. is really none of the tips to get bridge or transparent networking like in https://forums.docker.com/t/bridge-with-docker-for-windows/30936 working?

guanggevwu, 2023-12-08:
I have the exact same issue for running Basler ethernet camera in wsl.
The connection is fine and can get parameter information but can't grab image.
Any update?

DxTa, 2025-04-07:
Hi guys, just to check if Pylon Viewer inside WSL is working for you? When opening the application, I managed to connect to the camera (not able to auto-scan but can connect with an IP). However, when start capturing, it just show blank screen. I'm not so sure if this is Pylon Viewer or WSL GUI app issue. Hence, any suggested direction is appreciated.

[image]

SMA2016a, 2025-04-07:
Most properly, the destination address is not correct on the streaming parameter.

DxTa, 2025-04-08:
@SMA2016a can you help elaborate more? I see you [mentioning ](https://github.com/basler/pypylon/issues/581)able to make GigE work in WSL setting, so maybe I can learn from your setup.

It seems from Pylon Viewer WSL, I managed to connect and see parameters of the camera, but not showing camera frame yet.

DxTa, 2025-04-15:
I managed to get the camera frame from Pylon Viewer (WSL) By create a special Virtual Switch using "Hyper-V manager". After that, the cameras and the WSL can be in the same network, setting the camera would then make it working as expected.

I believe then pypylon should work normally (as well as Pylon Gstreamer SDK).

DxTa, 2025-10-30:
I tested further with more cameras and ends up write a windows excutable file based on pypylon sdk. Then forward the frame to stream inside WSL.

From my view point, this separation seems more reliable.

Anyhow @nische , mirrored network is not working for me as well. You should create a bridge network in Hyper-V Manager so that your wsl is in the same subnet with your cameras. This works for me, but I lost internet connection and not yet able to make it work yet. If you are interested in this approach, maybe we can chat more and you can help check it further if you can work through it? My email: duc-at-reversely.fi

SMA2016a, 2025-10-31:
> [@SMA2016a](https://github.com/SMA2016a) can you help elaborate more? I see you [mentioning ](https://github.com/basler/pypylon/issues/581)able to make GigE work in WSL setting, so maybe I can learn from your setup.
> 
> It seems from Pylon Viewer WSL, I managed to connect and see parameters of the camera, but not showing camera frame yet.
> 
> p/s: I'm just checking if Pylon Viewer can do that, and maybe will try to dive to linux container later. This is a bummer when I need to support WSL for edge-windows machine :|. However, if this is not working, I'm thinking of having a windows service to forward frame to the linux containers/WSL

sure, 

-     announce the camera’s IP address to the pylon manually
-     set a fixed streaming port for the pylon stream grabber
-     set the destination IP in pylon
-     enable port forwarding on WSL host system

or 

typein in power shell 

Enable-WindowsOptionalFeature -Online -FeatureName Microsoft-Hyper-V -All