basler/pypylon #572: GigE camera grab timeout using pypylon

thedim-witted, 2022-12-15:
Hi, I am using Ubuntu 20 where I am able to see and capture frames from my acA1920-40gc camera using the Pylon Viewer but I am getting the below error when I am trying to capture it using a python script and pypylon.

__Traceback (most recent call last):
  File "test_gige.py", line 28, in <module>
    res = cam.GrabOne(10000)
  File "/usr/local/lib/python3.8/dist-packages/pypylon/pylon.py", line 3604, in GrabOne
    return _pylon.InstantCamera_GrabOne(self, *args)
_genicam.TimeoutException: Grab timed out. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1362)
__

This is the code used to grab:

`from pypylon import pylon
import cv2

#ip_address = '192.168.1.40'
#info = py.DeviceInfo()
#info.SetPropertyValue('IpAddress', ip_address)
#
#camera = py.InstantCamera(py.TlFactory.GetInstance().CreateFirstDevice(info))
#
#tlf = py.TlFactory.GetInstance()
#camera =  py.InstantCamera(tlf.CreateFirstDevice())
#camera.Open()
#camera.StreamGrabber.DestinationPort = 50100
#res = cam.GrabOne(1000)

tlf = pylon.TlFactory.GetInstance()
tl = tlf.CreateTl('BaslerGigE')
cam_info = tl.CreateDeviceInfo()
cam_info.SetIpAddress('192.168.1.40')
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
#cam.StreamGrabber.DestinationPort = 50100

#print(cam.getModelname())
#print(cam.getMacAddress())

cam.Open()
#cam.StreamGrabber.DestinationPort = 50100
res = cam.GrabOne(10000)

print(res)`

Please help in resolving this.

thiesmoeller, 2022-12-15:
do you also set the IP address  in pylon viewer manually?
If not.. [grab.py](https://github.com/basler/pypylon/blob/master/samples/grab.py) should just work

SMA2016a, 2022-12-16:
can you ping your camera?

thedim-witted, 2022-12-16:
@thiesmoeller Yeah. We have to set the IP address so that we access the same camera every time
@SMA2016a  acA1920-40gc

SMA2016a, 2022-12-17:
do you use "Add remote GigE Camera" in Pylon viewer?  

Would you try this?

'''
tlf = pylon.TlFactory.GetInstance()
tl = tlf.CreateTl('BaslerGigE')     # remove this line
cam_info = tl.CreateDeviceInfo()  #release this with following line
cam_info = pylon.DeviceInfo()

cam_info.SetIpAddress('192.168.1.40')
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))

'''

hutomosaleh, 2022-12-22:
I'm also experiencing the same issue.

Ubuntu 20.04
a2A5328-4gcPRO

Here is my code snippet
```
from pypylon import pylon

ip_address = '192.168.0.4'
tlf = pylon.TlFactory.GetInstance()
tl = tlf.CreateTl('BaslerGigE')
cam_info = tl.CreateDeviceInfo()
cam_info.SetIpAddress(ip_address)
camera = pylon.InstantCamera(tlf.CreateDevice(cam_info))
cam_info = camera.GetDeviceInfo()
print(
    f"Name: {cam_info.GetModelName()}",
    f"IP: {cam_info.GetIpAddress()}",
    f"Mac: {cam_info.GetMacAddress()}"
)

camera.Open()
print("Camera opened")
camera.GrabOne(1000)
camera.Close()
```

It shows the correct model and mac address.
It works fine in Pylon Viewer as well as in my C++ code.
What could be the problem?

SMA2016a, 2022-12-23:
@hutomosaleh  have you tried the code that I have posted?

bsteenput, 2023-02-08:
Hello @SMA2016a 

I'm experiencing the same problem but I'm not sure to understand what you are asking to test.
I'm executing the code from a virtual machine in another network but I'm able to ping the camera. 
This is the code I'm executing:

```python
from pypylon import pylon

ip_address = '10.1.200.13'

factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info =ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)
camera.Open()

numberOfImagesToGrab = 100
camera.StartGrabbingMax(numberOfImagesToGrab)

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data.
        print("SizeX: ", grabResult.Width)
        print("SizeY: ", grabResult.Height)
        img = grabResult.Array
        print("Gray value of first pixel: ", img[0, 0])

    grabResult.Release()
camera.Close()
```

I'm getting the following error:

```
  return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1037)
```

JoaqBS, 2023-02-23:
Hello @bsteenput 
Did you manage to solve this issue?

SMA2016a, 2023-02-23:
remove this line

ptl = factory.CreateTl('BaslerGigE')

bsteenput, 2023-03-10:
Hi @JoaqBS ,

I am currently using a camera that is located on a different network than my machine. In order to establish a connection, I need to be able to specify the camera's IP address. Interestingly, I have noticed that the same code works flawlessly on a machine that is on the same network as the camera, and even the pylon viewer is able to retrieve pictures.

Upon investigating the issue using Wireshark, I found that while the camera is receiving all instructions and is able to send back its configuration, it is sending the UDP data stream back to 0.0.0.0.

I reached out to Basler support, and they recommended using CBaslerUniversalInstantCamera. I tried out the Grab_MultiCast.cpp sample code in the pylon SDK and tested the feature, and it works perfectly.

My question now is: is it possible to achieve the same functionality in pypylon?

SMA2016a, 2023-03-17:
set camera to multicast mode and then set the destination address. 

cam.StreamGrabber.TransmissionType ="Multicast"
cam.StreamGrabber.DestinationAddr ="IP of your target PC"

SMA2016a, 2023-06-13:
I have already posted a possible solution, that is 

ip_address = '192.168.1.40'

tlf = pylon.TlFactory.GetInstance()
cam_info = pylon.DeviceInfo()   
cam_info.SetIpAddress(ip_address)
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))

Simply use the universal  device info object instead of GigE one.

Raistlin707, 2023-06-14:
Hi @SMA2016a 

Thanks for your help! When I tried your solution eariler, i forgot to replace "CreateDeviceInfo" into "DeviceInfo" which resulted into an Exception. Your solution is indeed way better than the one I posted yesterday.

SMA2016a, 2023-06-14:
Thanks for the update

danyhar, 2023-06-21:
Hi, 

I came to this thread because I get an exception when I want to grab an image from a Basler acA1300, although I am able to find the camera using the depreciated Device object. It throws the following exception:
` Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1037)!`

So I decided to change to my code to the one posted above unfortunately now I get an Error that now device is available: 
`_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)` 

I am using a docker container (python 3.10) with pypylon installed. The camera got an static IP and I started the container with `--network:host` and mapped `/dev:/dev`. Between the device running the container and the camera is a PoE switch to power this camera and several others.

Now I'm confused if this is an Docker problem or if this is something with my code/setup or do I have to set specific parameters on the camera to get image to the container.

deanrobertcook, 2024-01-29:
Hi, thanks to all for the clues on solving the issue here. 

I had to do the following:

```python
tlf = pylon.TlFactory.GetInstance()
tlf.EnumerateDevices() # Note this additional line to @SMA2016a's answer above
cam_info = pylon.DeviceInfo()
cam_info.SetIpAddress(ip_address)
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
```

And then it worked. 

Without enumerating devices, I got the error `No device is available or no device contains the provided device info properties.`

I would like to get to the bottom of why exactly, if anyone can enlighten me here.

Cheers!